me-no-dev/EspExceptionDecoder #1: xtensa file missing

eumb, 2016-01-30:
Whenever I want to use the tool I receive the following message:

"ERROR: xtensa-lx106-elf-addr2line.exe not found!"

Where is the tool looking for it because mine is in : \AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\

me-no-dev, 2016-01-30:
should be \AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf\bin\

eumb, 2016-01-30:
I did reinstalled the stable arduino esp branch and still the same error. How could I make it that the error is more verbose?

me-no-dev, 2016-01-30:
Can you tell me your installation steps so I can repeat them exactly on a clean VM and see what might go wrong where.

eumb, 2016-01-30:
sure, I just copied the .jar from your latest EspExceptionDecoder.jar release into my workspace directory under: F:\10.ESP8266\Workspace\tools\EspExceptionDecoder\tool\, started my IDE and I have the ESP Exception Decoder under the Tools menu. Whenever I click on it , I have the error:
"ERROR: xtensa-lx106-elf-addr2line.exe not found!"  

I would like to mention that I am able to compile and upload ESP related sketches based on the staging or the stable version. 

 Directory of c:\Users\Yoyo\AppData\Local\Arduino15\packages\esp8266\tools

30-Jan-16  10:21 PM    <DIR>          .
30-Jan-16  10:21 PM    <DIR>          ..
30-Jan-16  10:21 PM    <DIR>          esptool
30-Jan-16  10:21 PM    <DIR>          mkspiffs
30-Jan-16  10:17 PM    <DIR>          xtensa-lx106-elf
30-Jan-16  10:21 PM    <DIR>          xtensa-lx106-elf-gcc

 Directory of c:\Users\Yoyo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf

30-Jan-16  10:17 PM    <DIR>          .
30-Jan-16  10:17 PM    <DIR>          ..
18-May-15  09:56 PM    <DIR>          bin
18-May-15  09:56 PM    <DIR>          lib
18-May-15  09:56 PM    <DIR>          libexec
18-May-15  09:56 PM    <DIR>          share
18-May-15  09:56 PM    <DIR>          xtensa-lx106-elf
               0 File(s)              0 bytes
               7 Dir(s)   7,177,289,728 bytes free

c:\Users\Yoyo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf>cd bin

c:\Users\Yoyo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf\bin>dir
 Volume in drive C has no label.
 Volume Serial Number is 6AE0-C54F

 Directory of c:\Users\Yoyo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf\bin

18-May-15  09:56 PM    <DIR>          .
18-May-15  09:56 PM    <DIR>          ..
19-Apr-15  02:00 PM         1,241,889 libiconv-2.dll
19-Apr-15  02:00 PM         5,101,699 xtensa-lx106-elf-addr2line.exe
.....
....

me-no-dev, 2016-01-31:
OK I was able to reproduce and fix this :)
https://github.com/me-no-dev/EspExceptionDecoder/releases/download/1.0.1/EspExceptionDecoder-1.0.1.zip

eumb, 2016-01-31:
Cool, it works now.
Thanks

closed 2016-02-04 by me-no-dev