me-no-dev/EspExceptionDecoder #2: Error file not found..

juiplus, 2016-02-04:
Whenever I want to use the tools I got the following message (using Decoder V1.0.1, Arduino 1.6.5 under Win8):
ERROR: abcd.ino.elf not found!
The reason for that is that the file is named abcd.cpp.elf! When I rename the file then the tool is running.. 
Please change:
elf = new File(getBuildFolderPath(editor.getSketch()), editor.getSketch().getName() + ".ino.elf");
to:
elf = new File(getBuildFolderPath(editor.getSketch()), editor.getSketch().getName() + ".cpp.elf");
Is that correct?

me-no-dev, 2016-02-04:
did they rename it recently then? I tested on one of the latest builds (Windows too) and elf name had ino before that... will add the extra check no problem :)

me-no-dev, 2016-02-04:
can you try this? https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/1.0.2

juiplus, 2016-02-04:
Sorry I can not tell if they renamed the file.. an or check for 'cpp' would be great :)
Thanks

juiplus, 2016-02-04:
Solved with 1.0.2 :) thank you!

closed 2016-02-04 by juiplus