me-no-dev/EspExceptionDecoder #3: [Feature request] Add Exception code info

av1024, 2016-02-20:
It will be cool if 'Exception(XX)' text would be decoded using for example [this](https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md) table (at least cause name and description)

me-no-dev, 2016-02-23:
should be visible in the trace itself? is it not? that is the reason I did not put it there

av1024, 2016-02-23:
There is only "Exception (28)" (for example) in trace. Sometimes with short exception name.

I am not very familiar with Java so I made monkey-patched version for my cubietruck's arduino 1.6.5 with java 1.7 (git code does not compiles with error near FileUtils usage): https://gist.github.com/av1024/018ffd57fae7e7f9d353

This is almost exactly what I want. As an perfect solution I want any suggestion like "Hey! Check PSTR/F/etc usage" for this exception ;)

me-no-dev, 2016-04-17:
I don't know how I forgot about this...
I looked through your code and it looks good but I know that there are gaps in the exception cause table or at least there are on the ESP32. Also maybe we should check if they are the same and use a two dimensional array to map the exception number with the exception string.
Where did you get your exception table from?

me-no-dev, 2016-04-17:
seems I read it with my ass... nevermind :) will merge it to the current source

av1024, 2016-04-18:
:))
BTW, is the ESP32 codes same as ESP8266? May be you should check what chip used?...

me-no-dev, 2016-04-18:
it's not exactly the same because the hardware is a bit different and it's running RTOS but it's on the same architecture :) you can check my ESP31B repo which is the arduino env for the new chip.
We have lots of headers there :)

av1024, 2016-04-18:
Looks promising!
But... I have 8 of 10 [PCB's](https://bitbucket.org/repo/9GdA6p/images/3496538974-board.jpg) for ESP8266 unsoldered so ESP31B not for me at least for half of year ))

Hmm.. Thinking out loud. When I wrote this patch I wanted to write a parser for wiki page... May be an "external" (local file or web link?) message source will be a solution? This will be helpful for updating and (for example) translation.

me-no-dev, 2016-04-18:
that is actually not a bad idea! I will see what options do I have to access local files for the tool. Maybe there is an option for translation...

me-no-dev, 2016-04-18:
BTW it looks like the exception numbers are the same for both chips, so we can go with single table for now.

me-no-dev, 2016-11-17:
fixed in the latest release :)

closed 2016-11-17 by me-no-dev