me-no-dev/EspExceptionDecoder #6: alternate build.path

pgScorpio, 2016-05-30:
This tool does not work if you set an alternate build.path in preferences.txt

ERROR: neither _sketchname_.ino.elf or _sketchname_.cpp.elf were found!
Did you forget to compile the sketch?

me-no-dev, 2016-05-30:
yes, the calculated build path is not actually exposed to the plugin, so we are trying to guess it as good as we can (for standard config that is).

electronicsguy, 2016-08-21:
I had this error as well. Could you add something to the error like "Also remove custom build-path if it exists" so that people can understand the cause?

electronicsguy, 2016-09-09:
Can't it read the arduino preferences.txt file and read the "build.path" string?

xetra11, 2016-09-24:
I have the same error when uploading my stuff via makefile. My sketchfolder has a folder called "build-esp8266" but when I open this sketch through the Arduino IDE the Decoder Tool doesn't realize that its available. Also when I copy the .elf to the root of the sketchfolder it doesnt work. Where do I need to place this file so the Tool will be able to read it?

doleynikov, 2016-09-29:
I have same problem. I defined build.path to preferevces.txt? so all .elf files are in same folder. Where I should put all those *.elf to get this important tool working?

Thank you.

doleynikov, 2016-09-29:
or, at least , let me choose correct .elf from somekind of menu.

me-no-dev, 2016-09-29:
I have all of my time invested into getting the initial platform for ESP32 ready. Once that is done, I can spend the time to implement support for it in the decoder along with some/all of the requested features :)

xetra11, 2016-10-01:
@me-no-dev 
Thanks buddy for taking your time at all! Hopefully in the near future somebody is able to help you out better...maybe me :P

me-no-dev, 2016-11-17:
FIXED! if elf is not found a dialog will be shown to manually locate it
please download latest release

closed 2016-11-17 by me-no-dev