me-no-dev/EspExceptionDecoder #7: Allow to decode stack for binaries at any location

Anton-V-K, 2016-06-23:
The tool works fine when decoding stack trace for ELF/HEX at the Arduino's build directory.
In some cases the binaries may be old (i.e. not freshly built), and they are located at other directory.

Use case:
1. Build a sketch and deploy it to a device.
2. Continue development (i.e. modify the code).
3. Receive crash report from a user for the deployed version.
4. Feed old HEX/ELF to the decoder along with the stack trace.

At the moment the step 4 should be preceded with extra actions: get corresponding sources from VCS and build them.

me-no-dev, 2016-06-23:
I will take a look at how to trigger file open dialog, if that is possible :)

me-no-dev, 2016-11-17:
FIXED! if elf is not found a dialog will be shown to manually locate it
please download latest release

closed 2016-11-17 by me-no-dev