me-no-dev/EspExceptionDecoder #8: Allow EspExceptionDecoder to work from the CLI.

skorokithakis, 2016-08-13:
This may be related to #7. I don't use the Arduino IDE, so it would be great if I could just run:

~~~
java -jar ExceptionDecoder.jar <myfile.elf> <mystack.txt>
~~~

And get the decoded exception back.

boraozgen, 2016-08-31:
It would be great to have this feature. I always replace the binaries compiled by the Arduino and with the ones compiled by the makefile and so on...

littleyoda, 2016-10-03:
I created a CLI-Version:
https://github.com/littleyoda/EspStrackTraceDecoder

me-no-dev, 2016-11-17:
@littleyoda you can change your array with exceptions with the one I just added to this repo :) gives better explanation.
Also you might want to make a PR to this repo's README and add link and short usage for your tool :)

me-no-dev, 2016-11-17:
@ivankravets could maybe use it in platformio to further extend support for ESP?

littleyoda, 2016-11-19:
@me-no-dev  Thanks for the new description of the exceptions. I've added them to my tool.

me-no-dev, 2016-11-19:
@littleyoda will you make a pull request with some text for this repo's README so people can easily find your tool and I can close this issue?

littleyoda, 2016-11-19:
PR created

ivankravets, 2016-11-19:
@me-no-dev I've jsut made quick look for [https://github.com/me-no-dev/EspExceptionDecoder/blob/master/src/EspExceptionDecoder.java](/EspExceptionDecoder.java). Did you use Java especially for Arduino IDE? Would be good to have cross-platform tool where java is not installed. 
I think that this code coudl be easy ported to Python or C, where we can reuse it later on different architectures/OSs.

me-no-dev, 2016-11-19:
@ivankravets yes :) It's arduino-ide plugin. Surely a python version could be made :) It just parses the given dump and extracts addresses pointing to code that then gives to addr2line. I imagine platformio was the reason @skorokithakis asked for this feature

skorokithakis, 2016-11-19:
Yep, it was!

skorokithakis, 2016-11-19:
I have this one liner from the Gitter channel:

~~~
rm -f stack.txt && vi stack.txt && awk '/>>>stack>>>/{flag=1;next}/<<<stack<<</{flag=0}flag' stack.txt | awk -e '{ OFS="\n"; $1=""; print }' | <path_to_tools_bin>/xtensa-lx106-elf-addr2line -aipfC -e <path_to_elf> | grep -v "?? ??:0" ; rm -f stack.txt
~~~

janLo, 2018-01-16:
I just had the same problem and created https://github.com/janLo/EspArduinoExceptionDecoder