me-no-dev/EspExceptionDecoder #13: Not work at a new IDE 1.6.12

pauloeduardosp, 2016-10-06:
After a upgrade to new ide 1.6.12 decoder stop to work.
When I will open a decoder windown I receive this msg

ERROR: neither Varanda_Des.ino.elf or Varanda_Des.cpp.elf were found!
Did you forget to compile the sketch?

But i already compile and make a sketch upload

thiagodasilva, 2016-10-07:
I also ran into this issue, but I found a work around for now...

Arduino builds your sketch at a location like /tmp/arduino_build_XXXX
EspExceptionDecoder creates a directory at /tmp/build<somehash>.tmp
It is looking for those files in that directory, so copy all the files from the /tmp/arduino_build_XXX to the temp EspExceptionDecoder directory and you should be able to open the Exception window

pauloeduardosp, 2016-10-10:
tks for the answer,
i'll try and send you a feedback.

electronicsguy, 2016-10-22:
Yup, doesn't work OOTB with the new IDE. Hopefully it'll be fixed soon. 

btw, in Windows, the correct paths are:
copy all files from "C:\Users\<username>\AppData\Local\Temp\arduino_build_NNNN\" to "C:\Users\<username>\AppData\Local\Temp\buildNNNNN.tmp\"

Where NNNN are random numbers.

danielcbit, 2016-11-02:
can some one point to the correct paths at Mac OS?
Tried to create a /tmp/build.tmp folder and copy the build contents from /var/folders/8l/<hash>/T/arduino_build_NNNN.
Also tried to create a arduino_build_NNNN folder at same path where the Arduino IDE is building and copy this same folder to tmp.

Still no success...

mph070770, 2016-11-15:
I couldn't get the workaround to work. What version of IDE should I revert to if I want to get this working?

gfoiani, 2016-11-17:
Every time you restart Arduino IDE it builds the sketch in a folder with a different build number:
i.e. `/var/folders/8s/6k3r4rt57r38rl8r4ngd6j780000gn/T/arduino_build_XXXXX/`

I can get it work on Mac OS by linking the built sketch in the folder where the Exception Decoder looks for it, in this way:

`ln -s /var/folders/8s/6k3r4rt57r38rl8r4ngd6j780000gn/T/arduino_build_XXXXX/MySketch.ino.elf /var/folders/8s/6k3r4rt57r38rl8r4ngd6j780000gn/T/build9d362461b273ac7cd349f8fd844f1118.tmp/MySketch.ino.elf`

where XXXXX is the build number that changes every time you restart Arduino IDE. So you have to run this command every time you restart Arduino IDE checking in advance which is the new build number.

me-no-dev, 2016-11-17:
FIXED!
please download latest release

closed 2016-11-17 by me-no-dev

pickeld, 2016-11-18:
hey . i did download the latest release. still get the error. 
ERROR: neither testjason.ino.elf or testjason.cpp.elf were found!
Did you forget to compile the sketch?

me-no-dev, 2016-11-18:
not possible :) did you restart the IDE? which version did you download?

pickeld, 2016-11-18:
im using Arduino IDE Version 1.6.12
and EspExeptionDecoder Version 1.0.6
Ive extracted the jar file into this location:
C:\Users\David\Documents\Arduino\tools\EspExceptionDecoder\tool\EspExceptionDecoder.jar
restarted the Arduino IDE opened the SerialMonitor, uploaded my sketch, got the error on the serial monitor. then when i pressed ESP Exception Decder i see on screen 

`Uploading 233344 bytes from C:\Users\David\AppData\Local\Temp\arduino_build_182492/testjson.ino.bin to flash at 0x00000000
................................................................................ [ 35% ]
................................................................................ [ 70% ]
....................................................................             [ 100% ]
ERROR: neither testjson.ino.elf or testjson.cpp.elf were found!
Did you forget to compile the sketch?`

pickeld, 2016-11-18:
im so sorry, i kept an old version on other location . seems like i saved the new one in the wrong location.
this issue is Solved.
thank you for the hard work!

me-no-dev, 2016-11-19:
glad it works :)