me-no-dev/EspExceptionDecoder #17: Not Starting.

engrziarehman, 2016-12-01:
I am using arduino 1.6.9 IDE, whenever i try to use EspExceptionDecoder i get following error:
`Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.ficeto.esp.EspExceptionDecoder.getBuildFolderPath(EspExceptionDecoder.java:160)
	at com.ficeto.esp.EspExceptionDecoder.createAndUpload(EspExceptionDecoder.java:262)
	at com.ficeto.esp.EspExceptionDecoder.run(EspExceptionDecoder.java:344)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`

me-no-dev, 2016-12-01:
update your IDE or use older version. Latest is for newer IDEs

me-no-dev, 2016-12-01:
https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/1.0.4

WilbertE, 2016-12-01:
I also get this issue. What Arduino IDE version is needed? I have 1.6.9

engrziarehman, 2016-12-02:
Ok i have used 1.6.13, it worked. thanks.!!!