me-no-dev/EspExceptionDecoder #18: Select ELF dialog on start

eketjall, 2016-12-02:
When trying to open the tool I get a file open dialog asking me to select an elf-file. 
What file am I supposed to select?

Latest release 1.0.6 and IDE 1.6.13

me-no-dev, 2017-01-02:
Dialog will not show if you have compiled the sketch :) its there in cases where you have custom build output or you want to manually pick the elf that should be decoded against. If you compile/upload the sketch before opening the plugin, elf will be autoselected and dialog will not show.

sethten, 2017-03-10:
Thanks! I had uploaded a sketch, opened the serial monitor and saw the exception. Then I followed the instructions in the readme here to add this tool. I restarted the Arduino IDE, opened the serial monitor again and copied the output. Then I started the tool and it asked me to select a .ELF file. But, now I know I just needed to compile again after restarting the IDE. It worked and was very helpful!

Defozo, 2017-05-24:
Worth mentioning on the main readme perhaps?

seth10, 2017-05-25:
I do not remember much as to what I was working with. If this is an issue many new users will run in to, I think a clear notice would be helpful.

delixfe, 2017-06-09:
I run into the same problem. Looking at the Github issues fixed it within minutes for me.

electriccord, 2017-08-21:
Same problem.  Even though I had not changed the code since downloading it, a restart of the IDE lost the file and asked for an ELF.  That would have been good to know before waiting hours for the exception to reoccur.