me-no-dev/EspExceptionDecoder #19: Decode Failed

fabltd, 2016-12-06:
The following decode fails. 

Not sure whats wrong.

ctx: cont 
sp: 3ffef290 end: 3ffef530 offset: 01b0

>>>stack>>>
3ffef440:  00000070 3ffee404 3ffee398 402020dd  
3ffef450:  00000024 3ffee404 3ffee398 40202091  
3ffef460:  3ffef4f0 00001d14 3ffee398 402026f0  
3ffef470:  0000003d 00001d14 3ffee398 40202750  
3ffef480:  3ffef4f0 3ffee398 3ffee398 4020296c  
3ffef490:  0000000e 00000001 00000001 40201f28  
3ffef4a0:  3fffdad0 0000000e 3ffee4e0 00001cec  
3ffef4b0:  00000001 3ffee398 00000002 4020200d  
3ffef4c0:  3ffef4f0 0000000e 00000002 00000028  
3ffef4d0:  40201136 000003e8 3ffee4e0 3ffee504  
3ffef4e0:  3fffdad0 3ffee398 3ffee4e0 40201e10  
3ffef4f0:  6c6c6548 6f57206f 20646c72 00003223  
3ffef500:  00000000 00000000 00000001 40203319  
3ffef510:  3fffdad0 00000000 3ffee4fc 40203344  
3ffef520:  feefeffe feefeffe 3ffee510 40100718  
<<<stack<<<

me-no-dev, 2016-12-06:
Seems OK. What is the output?

fabltd, 2016-12-06:
No decode

For me:-(

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661

fabltd, 2016-12-06:
Also whys it keep asking for me to open a file?

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661

me-no-dev, 2016-12-06:
maybe you should have started with that ;)
the plugin requires the ELF file that was used to make the binary that you uploaded. If you compile the sketch that you want to debug and then click on the plugin, it will decode the addresses from that ELF, else you need to show it where to find it (in case that you have some custom build procedure)

fabltd, 2016-12-06:
Hi

Where does it store the elf files ?

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661

me-no-dev, 2016-12-06:
in the build folder that Arduino changes every time you open a sketch. It's not a constant place

fabltd, 2016-12-06:
How do I find it?

Regards,
Jeremy Turner
Managing Director
Jeremy@fabcontrols.co.uk
Mobile 0797 131 4661

me-no-dev, 2016-12-06:
you do not need to find it. If you have build the sketch, the plugin will find it for you. If the plugin does not find it, then it's on some custom place that neither me, nor the plugin can tell where it is :)

pherris, 2017-01-07:
@fabltd the file prompt should only be shown if you have not compiled & uploaded your sketch to your device _while the IDE is running_. In that case, the tool cannot find the `.elf` file, therefore it cannot handle the stack trace. To solve this:

- close/open the Arduino IDE
- upload the sketch
- open the serial monitor and capture your stack trace:
```
>>>stack>>>
...just the stuff in here
<<<stack<<<
```
- open the Decoder from the tools menu (you should not be prompted for the file)
- paste the stack trace FTW!

@me-no-dev please correct any nuances that are wrong here. If you'd like I can submit a PR to the README.

laercionit, 2018-05-25:
Decode Failed 

Core 2.4.1
Arduino IDE 1.8.5
Board NodeMCU 12-E 1.0

Example:
Soft WDT reset

ctx: cont 
sp: 3fff32a0 end: 3fff3490 offset: 01b0

>>>stack>>>
3fff3450:  3fffdad0 00000000 3fff23a0 4021de93  
3fff3460:  00000000 00000000 00000000 feefeffe  
3fff3470:  feefeffe feefeffe 3fff2458 402315e4  
3fff3480:  feefeffe feefeffe 3fff2470 40100710  
<<<stack<<<

tmorford, 2018-07-11:
I am having the same issues I can not decode ever not sure why I always get the same error  "Decode Failed". Has anyone fixed this yet?

laercionit, 2018-07-11:
Yes,...

Core 2.4.1
Decoder 1.06

Just did not work with the decoder in the latest version.

tmorford, 2018-07-11:
That did work.. thank you very much!

laercionit, 2018-07-11:
@tmorford , 

It worked?

tmorford, 2018-07-11:
Yes it did thank you.