me-no-dev/EspExceptionDecoder #20: ERROR: xtensa-lx106-elf-addr2line.exe not found!

disneysw, 2017-01-01:
I have installed the jar and restarted the IDE. The exception decoder appears in the menu but I get a "ERROR: xtensa-lx106-elf-addr2line.exe not found!" when I try to start it. 

I can build and download esp8266 apps and the xtensa-lx106-elf-addr2line.exe is in the esp tools directory (under appdata). I am using Win10 and have tried with IDE v1.6.9 and the latest v1.8.0.

I also tried copying xtensa-lx106-elf-addr2line.exe into the same directory as the jar without sucess - any thoughts?

me-no-dev, 2017-01-01:
Did you install esp8266 through package manager ot git?

disneysw, 2017-01-01:
Package manager

disneysw, 2017-01-01:
But I did install the staging version as I need its features....

me-no-dev, 2017-01-01:
It should not matter. It should have worked in any of the cases. Ill see to give it a go and try to reproduce

pat1, 2018-02-12:
The same on Linux: