me-no-dev/EspExceptionDecoder #21: v 1.0.6 won't run in Arduino 1.6.11 and up-to-date Java

matasondrak, 2017-01-24:
This release gives following output in Arduino console:

`Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.ficeto.esp.EspExceptionDecoder.getBuildFolderPath(EspExceptionDecoder.java:160)
	at com.ficeto.esp.EspExceptionDecoder.createAndUpload(EspExceptionDecoder.java:262)
	at com.ficeto.esp.EspExceptionDecoder.run(EspExceptionDecoder.java:344)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`

Tested on Windows 7 x64 SP1, Arduino 1.6.11, ESP Exception Decoder 1.0.6, Java 8, build 11.

**Decoder version 1.0.2 works just fine.**

me-no-dev, 2017-01-24:
there is something wrong on your side. You are running the wrong Arduino or something. There is no way 1.0.2 is working on recent IDE ;) The change above was committed from the Arduino team themselves :)

aMarcireau, 2017-02-07:
I face the same error with Arduino 1.6.4 and Fedora 25.

me-no-dev, 2017-02-08:
@aMarcireau latest version is for IDE 1.6.11 and up

ddjikic, 2017-02-26:
I have the same issue on ubuntu , and arduino ide 1.6.5

`Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.ficeto.esp.EspExceptionDecoder.getBuildFolderPath(EspExceptionDecoder.java:160)
	at com.ficeto.esp.EspExceptionDecoder.createAndUpload(EspExceptionDecoder.java:262)
	at com.ficeto.esp.EspExceptionDecoder.run(EspExceptionDecoder.java:344)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`

slootsky, 2017-03-03:
same issue w/ Arduino 1.6.11 on Win10 x64.  Unsure of Java version.

Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.ficeto.esp.EspExceptionDecoder.getBuildFolderPath(EspExceptionDecoder.java:160)
	at com.ficeto.esp.EspExceptionDecoder.createAndUpload(EspExceptionDecoder.java:262)
	at com.ficeto.esp.EspExceptionDecoder.run(EspExceptionDecoder.java:344)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

rishikeshayre1310, 2017-03-03:
even i have same problem on arduino 1.6.8

Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.ficeto.esp.EspExceptionDecoder.getBuildFolderPath(EspExceptionDecoder.java:160)
	at com.ficeto.esp.EspExceptionDecoder.createAndUpload(EspExceptionDecoder.java:262)
	at com.ficeto.esp.EspExceptionDecoder.run(EspExceptionDecoder.java:344)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

me-no-dev, 2017-03-05:
Guys... latest version are made for the latest ArduinoIDEs ;) older version of ide-> older version of the plugin. Things in the IDE changed and we can not keep the plugin compatible with lower versions anymore.

xama5, 2017-08-14:
Using version 1.0.5 of this plugin for Arduino 1.6.4 worked for me.