me-no-dev/EspExceptionDecoder #22: I can't install this tool in my arduino IDE.

Joypacharya, 2017-03-15:
Hello 
i checked your tools details and follow step as per suggested but i can not install it.
So, help me install this tool.I really need this tool.
I am using Arduino ide  1.6.12

pfeerick, 2017-05-08:
Did you try reading the instructions on the front page? Specifically the second dot point where it says...
> Download the tool archive from releases page.
... with the word releases being a link to said page where you can download the zip file with compiled .jar file?

You need to download the zip file. Then create a folder in your Arduino sketchbook directory called tools. Then you need another folder inside that called EspExceptionDecoder. And another one inside that called tool. And the EspExceptionDecoder.jar goes in that final tool folder. Or if you just extract the zip file into your sketchbook\tools folder the other folders will be created for you.

shkhamd, 2017-12-16:
hi, I am trying to use in on windows 8 and arduino 1.6.9 but got the following error. 

`Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Sketch.getBuildPath()Ljava/io/File;
	at com.ficeto.esp.EspExceptionDecoder.getBuildFolderPath(EspExceptionDecoder.java:160)
	at com.ficeto.esp.EspExceptionDecoder.createAndUpload(EspExceptionDecoder.java:262)
	at com.ficeto.esp.EspExceptionDecoder.run(EspExceptionDecoder.java:344)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`

How do i fix this?

shkhamd, 2017-12-16:
version 1.0.5 worked for me. thanks.

microstronic, 2018-03-28:
execuse me..i use nodemcu v3  and  example sketch "Flash Led" in library Universal Telegrambot, but i tested in 10 minutes , the system reset and show me in serial monitor like that ...
i used arduino ide 1.8.5 and try to instal EspExceptionDecoder but, in ide arduino cannt show this tool

can everyone help me..

Exception (29):
epc1=0x4020e076 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0630 end: 3fff0be0 offset: 01a0

>>>stack>>>
3fff07d0:  00820000 00000082 3fff3d9c 4021aef2  
3fff07e0:  3ffefad4 3fff59b4 3fff3d9c 4021b0ec  
3fff07f0:  3fffba24 00000208 3fffba24 40100670  
3fff0800:  00000104 0000a910 00001522 00000042  
3fff0810:  00000040 00000082 3fff3fd4 4021b405  
3fff0820:  00800000 00000080 00000000 00000041  
3fff0830:  00000040 0000003f 3fff993c 00000041  
3fff0840:  3fffb60c 3fff3fd4 3fff993c 3fff6a0c  
3fff0850:  3fffbb20 ffffffff 3fff59b4 40206624  
3fff0860:  00000040 3fff59b4 3fff3fd4 3fff59b4  
3fff0870:  00000040 3fff59b4 3fff3fd4 4021bf3f  
3fff0880:  3fff545c 0000003f 7133b7a0 00000001  
3fff0890:  3fff59b4 3fff3fbc 3fff3fd4 00000001  
3fff08a0:  00000010 3fff59b4 0000000f 4021c194  
3fff08b0:  3fff993c 3fff1e9c 0000000f 00000001  
3fff08c0:  00000001 3fff1fcc 3fff20a1 4021af98  
3fff08d0:  00000100 3fff5d04 3fff209f 00000000  
3fff08e0:  00000100 3fff5d04 3fff209f 4021d31d  
3fff08f0:  3fff0920 00000000 000000d0 00000030  
3fff0900:  d3e09fcf 3fff1fd8 3fff9948 00000004  
3fff0910:  3fff2099 3fff4c2c 3fff2099 40218018  
3fff0920:  b2900303 473d1aa8 39e46d45 f1873ca9  
3fff0930:  aba177a8 1f9ed715 3558c5b7 087af8dc  
3fff0940:  4e4f4e17 70c8de58 60fec64b 9fcffe5a  
3fff0950:  3fff2099 00000004 3fff4c2c 402184ad  
3fff0960:  7a09c4cf 2757dde1 d361711d 4021571c  
3fff0970:  81ae9fb1 00000218 3fff90ac 40211281  
3fff0980:  00000000 00000004 00000004 402037b4  
3fff0990:  00000006 00001246 00000064 402065e4  
3fff09a0:  0000004a 00000000 00000026 00000004  
3fff09b0:  3fff2099 00000004 3fff4c2c 00000004  
3fff09c0:  3fff2099 00000004 3fff4c2c 40217e20  
3fff09d0:  5b1a59fa 00000bd3 00000bd3 00000027  
3fff09e0:  0000005c 3fff2099 3fff4c2c 402181fc  
3fff09f0:  00000010 3fff20cc 3fff4c2c 401004d8  
3fff0a00:  3fff0a58 0000000e 00000010 3fff66f4  
3fff0a10:  40203608 3fff4c2c 3fff0a50 01000000  
3fff0a20:  3fff0a58 000d149a 3fff4c2c 40217f88  
3fff0a30:  3fff4c2c 3fff6fac 3fff4914 3fff66f4  
3fff0a40:  3fff878c 000d149a 3fff4914 40203de9  
3fff0a50:  3fff4c2c 3fff6fac 4020656c 3ffefbc0  
3fff0a60:  00003a98 00003a98 00003a98 4020136f  
3fff0a70:  3ffef86c 00000000 3fff878c 3fff0b20  
3fff0a80:  000001bb 3ffe8dc0 3ffef86c 40204037  
3fff0a90:  000001bb 3ffef86c 3ffe8dc0 3fff0b20  
3fff0aa0:  000001bb 3ffef86c 3ffe8dc0 402040dd  
3fff0ab0:  401072a0 dca79a95 401072a0 dca79a95  
3fff0ac0:  3ffe8ab4 3fff0b6c 3ffef7f8 402042af  
3fff0ad0:  3fff5c9c 00000054 3fff0b20 4020614f  
3fff0ae0:  3ffefad4 00000106 3fff0b20 40206181  
3fff0af0:  3fff0b3c 3fff0b3c 3fff0b20 402061f0  
3fff0b00:  3fff0b3c 3fff0b3c 3ffef7f8 3ffefbb8  
3fff0b10:  3fff0b3c 3fff0b3c 3ffef7f8 40205853  
3fff0b20:  3fff1944 0000005f 00000054 00000100  
3fff0b30:  00000000 00000000 00000000 00000000  
3fff0b40:  00000000 00000000 00000000 00000000  
3fff0b50:  00000000 00000000 00000000 00000000  
3fff0b60:  00000000 0000000b 3ffe8b77 3fff5f3c  
3fff0b70:  0000000f 00000000 3fff5c9c 0000005f  
3fff0b80:  00000054 00000000 00000000 00000000  
3fff0b90:  00000000 00000001 00000000 00000000  
3fff0ba0:  feefeffe 3ffef88c 3ffefaf8 3ffefbb8  
3fff0bb0:  3ffef7f4 3ffef7f8 3ffefbb0 40202575  
3fff0bc0:  3fffdad0 00000000 3ffefbb0 402065b8  
3fff0bd0:  feefeffe feefeffe 3ffefbc0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset