me-no-dev/EspExceptionDecoder #23: Line numbers do not match source code

drschlaumeier, 2017-05-23:
Hallo,
I just installed EspExceptionDecoder in Arduino IDE 1.8.2 to debug a nasty crash in my Webserver.
However, I have the feeling that the line numbers shown in decoder do not match the real line numbers in the source file. For example:

`0x4020ea20: esp_yield at D:\data2\Programme\_Progamming\ArdinoIDE\arduino-1.8.2\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
`

If I search the source file line number 57 than:
```

54 int atexit(void (*func)()) {
55    (void) func;
56    return 0;
57 }
58 extern "C" void ets_update_cpu_frequency(int freqmhz);
59 void initVariant() __attribute__((weak));
60 void initVariant() {
61 }
62
63 extern void loop();
64 extern void setup();
65
66 void preloop_update_frequency() __attribute__((weak));
67 void preloop_update_frequency() {
68 #if defined(F_CPU) && (F_CPU == 160000000L)
69     REG_SET_BIT(0x3ff00014, BIT(0));
70    ets_update_cpu_frequency(160);
71 #endif
72 }
```

So, how to interprete the line numbers?

Thanks

me-no-dev, 2017-08-02:
they do not ;) and really there is no way to make them match as those are the line number of the actual generated source after all includes and defines have been taken care of. They do tell function and file and if you get a couple, could also hint you if they are above or below each other.

earlephilhower, 2018-01-23:
Resurrecting an old issue, @igrr has traced it down to something in the toolset beyond the preprocessor, unfortunately: https://github.com/jcmvbkbc/binutils-gdb-xtensa/issues/5