me-no-dev/EspExceptionDecoder #24: Not Supported on ESP32

g-dalli, 2017-08-02:
I started with wifi on ESP32. The libraries are probably mixed up with standard arduino wifi.h. I got an exception on core 1 error and tried this promising tool. After starting it tells me Not Supported on ESP32.
I use arduino ide 1.8.3. Restart of the ide or compiling/upload the Sketch again didn't fix it.
Thank you for your support!

me-no-dev, 2017-08-02:
you need newer ArduinoIDE and the latest version of this tool ;)

g-dalli, 2017-08-02:
sorry to ask again, but 1.8.3 is the current arduino IDE, 
1.0.6 is the current version of your tool, right?
I just started, so may be I'm wrong ;)

tedder, 2017-10-17:
I'm using 1.8.3 and it complains about not being supported on "arduino-esp32". I suspect the longer string is the problem in my case. Not sure where it sniffs that from.

chegewara, 2017-11-12:
The problem is arduino does not recognise esp32 hardware if it is installed in arduino-esp32, when you delete arduino from folder name and leave just esp32 all should works fine. It has something to do with hardware configuration files inside arduino-esp32

arnold-b, 2019-05-03:
Hello,,
i´m using Arduino 1.8.9. If i try to use your tool (release 1.1.0). 
I also get the error: not supported ESP32. 
But i think, its useless to write. Because the last comment is already 2017

chegewara, 2019-05-03:
Yes, it was my comment, but since very long time it is working with esp32 to me. If i remember i didnt follow installation instruction at first.