me-no-dev/EspExceptionDecoder #26: Requests ELF file instead of copy/paste stack trace

GadgetRho, 2017-08-22:
The readme file states that one should copy and paste the stack trace, but this tool wants an elf file.

Torben80, 2017-08-31:
...Compile the sketch and it will not ask for elf...
someone wrote a few days ago

MrUrdam, 2018-02-08:
This solved the problem for me.

dresende, 2018-02-21:
Does not always work. You can find the elf in `Libraries/Arduino15/packages/esp8266/hardware/esp8266/**version**/bootloaders/eboot/eboot.elf`.

pauloeduardosp, 2018-04-05:
i found the elf file, but i didn't know where i need to put then

dresende, 2018-04-05:
The elf file is needed when the ESP Exception Decoder window opens and requests it. If it just works you don't need to point to the elf file.

zuqualla, 2019-06-05:
> 
> 
> Does not always work. You can find the elf in `Libraries/Arduino15/packages/esp8266/hardware/esp8266/**version**/bootloaders/eboot/eboot.elf`.

This file is also good for esp32?

thataintme, 2019-10-31:
> ...Compile the sketch and it will not ask for elf...
> someone wrote a few days ago

Can you please tell me how to compile a .jar file? is that what a sketch is?

RickHolc, 2019-11-01:
After installing Arduino 1.8.10 and reinstalling the ESPExceptionDecoder, now each time I start it I am asked to locate the *.Elf file. Before this never happened. What config file contains the path to the Elf file. Can I update it so it never asks again?

nicechocolate, 2021-01-12:
Makes sense that recompiling is needed, once you think about what the decoder is trying to achieve and how it will do it.

I, too, hadn't realised that it would be the compiler that made the ELF that it needs.  Very helpful to know this.

SHERAMII, 2021-08-04:
> 
> 
> ...Compile the sketch and it will not ask for elf...
> someone wrote a few days ago

thanks this worked