me-no-dev/EspExceptionDecoder #27: no item in Arduin IDE tools to start  EspExceptionDecoder

kdbqlb, 2017-10-27:
I try to install the EspExceptionDecoder in the german Arduino IDE.
What ever Folder I use to unzip the files the tool menu remains empty. 
So I m looking for a detailed screeenshot showing the full path of the Folder an ALL files in the folder.
And please give a screenshot of prferences.txt ( Settings of the IDE )
Thank you

me-no-dev, 2017-10-30:
You do not need to touch anything other than extract into your [sketches folder]/tools.

kdbqlb, 2017-10-30:
I tried it on 2 Windows PCs but The Tool Menu remains empty. Therefore I ask for detailed path and screenshot showing all necessary files.
Thank you 
Klaus-Dieter 🚴🏊🏼⛵️⛷🎾😊

me-no-dev, 2017-10-30:
Not a Windows user, so my screenshots will not do any good to you. Maybe you post a screenshot of what you are trying to do?

kdbqlb, 2017-11-01:
Hi,

here are the screenshots of my “installation” of EspExceptionDecoder.

But there is no tool in the menu. Why ?

Thanks Klaus-Dieter

teprom, 2017-11-01:
Me neither, do not appear anything for me on Arduino IDE menu's options.

me-no-dev, 2017-11-01:
@kdbqlb  i do not see screenshots :)

BTW just did a clean install on my Mac and all is fine.
By any chance you guys are using portable version of the IDE?

teprom, 2017-11-01:
No, I'm using installed IDE ( PT language ). File "EspExceptionDecoder.jar" extracted to folder "C:\Arduino\tools\EspExceptionDecoder\tool". At Tools meny there isn't anything there. Should I reboot system aswell ?
[image]

teprom, 2017-11-02:
Solved!

I had 2 Arduino folders on my Laptop (I don't know why).
Once copied to the other one, now the option is available there.

Thanks.