me-no-dev/EspExceptionDecoder #32: ESPExceptionDecoder is not showing up Arduino 1.8.5 ESP Core 2.4.0

srankur, 2018-01-28:
Hi,
I have followed up the instruction "https://github.com/me-no-dev/EspExceptionDecoder". 

Still "Tools > ESP Exception Decoder" not showing up after even restart, 

My config as follows- 
OS- Ubuntu 17.10
IDE - Arduino 1.8.5
ESP Core - 2.4.0
Folder path. home/"user"/installations/Arduino/tools/EspExceptionDecoder/tool/EspExceptionDecoder.jar

Please help getting it resolve. Appreciate your help!! 
Thanks,
Ankur

arihantdaga, 2018-03-09:
you'll have to extract it into sketches directory. I think you are doing it in the arduino15(In m case mac os) or arduino installation directory maybe.. 
Go and extract this into [sketches_directory]/tools folder

laercionit, 2018-05-20:
Same problem, any help?

arihantdaga, 2018-05-23:
@laercionit  Did you extract in your sketchDirectory/tools folder ?

laercionit, 2018-05-23:
@arihantdaga 

Hi, thanks for trying to help.

$ ArduinoFolder \ tools \ EspExceptionDecoder \ tool \ EspExceptionDecoder.jar

The ArduinoIDE displays the item in the menu normally, Esp Exeption Decoder.

Unfortunately whenever I need to perform the decoder and put the return of the stack I get the msg of Decode Failed.

I am using the following software:
IDE Arduino 1.8.5
ESP8266 Core 2.4.1 (Installing with Boards Manager)
EspExceptionDecoder 1.1.0

Decode does not work in any situation, it always decode failed.

arihantdaga, 2018-05-23:
@laercionit  Can you paste your stack here what are trying to paste in exp exception decoder..  ?

laercionit, 2018-05-23:
>>>stack>>>
3fffff00:  3ffe84dc 0000000e 3ffeed88 40204e68  
3fffff10:  3ffe84dc 3ffeed88 3ffeed88 40204fb1  
3fffff20:  3ffe8aa0 40203680 00000000 40204fb1  
3fffff30:  3ffe8a80 3ffeed88 3ffeed88 40204fdc  
3fffff40:  3ffe84dc 3ffeed88 3ffeed88 40205028  
3fffff50:  3ffe84dc 3ffeed88 3ffee8d8 40202a35  
3fffff60:  40205910 00000000 000003e8 feefeffe  
3fffff70:  5245414c 2e4f4943 00006264 feefef00  
3fffff80:  3ffef82c feefeffe feefeffe feefeffe  
3fffff90:  00000001 feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3ffeedac  
3fffffb0:  3fffdad0 00000000 3ffeeda5 40205500  
3fffffc0:  feefeffe feefeffe 3ffe8504 40100739  
<<<stack<<<

laercionit, 2018-05-25:
@arihantdaga 

Some example:
>>>stack>>>
3fff3450:  3fffdad0 00000000 3fff23a0 4021de93  
3fff3460:  00000000 00000000 00000000 feefeffe  
3fff3470:  feefeffe feefeffe 3fff2458 402315e4  
3fff3480:  feefeffe feefeffe 3fff2470 40100710  
<<<stack<<<

laercionit, 2018-05-26:
How do I fix the Decoder problem locating the correct elf file without having to do this type of maneuver manually?

bha14ur, 2018-10-20:
I download the tool archive from Github. but i cann't find the tool directory in that archive. and i also cann't found this ".jar" file.

dresende, 2019-02-28:
On MacOS, you should put it on `~/Documents/Arduino/tools/`.

pana216, 2019-06-03:
> On MacOS, you should put it on `~/Documents/Arduino/tools/`.

This actually saved me thank you very much!

michaelradhuber, 2019-11-04:
> 
> 
> > On MacOS, you should put it on `~/Documents/Arduino/tools/`.
> 
> This actually saved me thank you very much!

....On Windows10 too, thanks for the hint!

flavio-fernandes, 2019-12-23:
> you'll have to extract it into sketches directory. I think you are doing it in the arduino15(In m case mac os) or arduino installation directory maybe..
> Go and extract this into [sketches_directory]/tools folder

Yes, that made it work. TY TY TY!!!!

flavio-fernandes, 2019-12-23:
> On MacOS, you should put it on `~/Documents/Arduino/tools/`.

As mentioned by @arihantdaga  check and see what the Arduino preferences have set as 
```
sketchbook.path=
```

Then, add a tools directory under that, like:
sketchbook.path=/Users/flaviof/projects_arduino

```
 (venv) OSX ~/projects_arduino
$ cd ~/projects_arduino
$ tree tools
tools
└── EspExceptionDecoder
    └── tool
        └── EspExceptionDecoder.jar
```

supuflounder, 2020-04-28:
I'm having the same problem with Arduino 1.8.12 (Windows Store 1.8.33.0)

I've attached two screenshots.  Note that after I installed it, I restarted the Arduino IDE.
[image]
[image]

supuflounder, 2020-04-28:
Ignore previous message.  I was off by two directory levels.  Fixed.

flavio-fernandes, 2020-04-28:
glad to hear. ;)

TrinetraM, 2020-12-03:
Hey I have done the same thing But Tool is not visible in my arduino
Please reply its Urgent
[image]

vibeeshanan, 2021-02-03:
@TrinetraM 
same problem I got.

vibeeshanan, 2021-02-03:
can anyone help?

TrinetraM, 2021-02-03:
Go to releases page and download .jar file and paste it in
tools/EspExceptionDecoder you most probably have downloaded the master copy
which includes sec folder and readme files even I did the same mistake

Trinetra

TrinetraM, 2021-02-03:
You need to click on the highlighted releases page link and download the
tool extract it to tools folder in your sketchbook
Trinetra

vibeeshanan, 2021-02-03:
@TrinetraM  No, what did you tell me about the above solution same I do, but it doesn't show the ESPExceptionDecoder in the tools section, what I do?

TrinetraM, 2021-02-03:
Can u send me the screenshot of your tools folder
Trinetra

vibeeshanan, 2021-02-03:
> I'm having the same problem with Arduino 1.8.12 (Windows Store 1.8.33.0)
> 
> I've attached two screenshots. Note that after I installed it, I restarted the Arduino IDE.
> [image]
> [image]

the same issue I am facing but it doesn't work how did you solve it?

TrinetraM, 2021-02-03:
Just extract the zipfile to tools folder. After extraction your directory
structure should look like this
Arduino/tools/EspExceptionDecoder/tool/EspExceptionDecoder.jar

Trinetra

vibeeshanan, 2021-02-03:
[image]
this my tool folder

vibeeshanan, 2021-02-03:
@TrinetraM can you give me any suggestions?

TrinetraM, 2021-02-03:
I will send you a zip file just extract that in Arduino/tools folder that
should fix your problem

Trinetra

vibeeshanan, 2021-02-03:
Ok I Will wait for your zip folder @TrinetraM

TrinetraM, 2021-02-03:
Can you give me your whatsapp number gmail is blocking it for some reason

Trinetra

TrinetraM, 2021-02-03:
Can you give me your whatsapp number gmail is blocking the zip file

Trinetra

vibeeshanan, 2021-02-03:
for security reasons, I can't share my mobile number is there any other way
to send me an option?

TrinetraM, 2021-02-04:
Skype id ?

Trinetra

TrinetraM, 2021-02-04:
Share me your email id I will send a Google drive link download the zip
file and extract it to Documents/Arduino/tools folder

Trinetra

TrinetraM, 2021-02-04:
EspExceptionDecoder-1.1.0 (1).zip
<https://drive.google.com/file/d/1fjg-5BEG-Z0Ay2NNgTTHHUIS8tFcY3YP/view?usp=drivesdk>

Trinetra

vibeeshanan, 2021-02-04:
Thank you so much, I'll check it out.

TrinetraM, 2021-02-04:
Did you install it ? Did it work ?

Trinetra

SunBeenMoon, 2021-05-06:
@TrinetraM  
Hi, I'm stuck in same problem. 
Can I get access the .zip file? 

Thanks

TrinetraM, 2021-05-06:
https://drive.google.com/file/d/1fjg-5BEG-Z0Ay2NNgTTHHUIS8tFcY3YP/view?usp=drivesdk

Trinetra

pmazariegos-bitmec, 2021-10-15:
> https://drive.google.com/file/d/1fjg-5BEG-Z0Ay2NNgTTHHUIS8tFcY3YP/view?usp=drivesdk Trinetra
> […](#)

Thanks TrinetraM for me it worked!!

daniel-ungureanu, 2022-12-29:
Hi! 

I have the same problem as it was described before. I placed the file in correct path, I checked also with the version share via Drive, but the tool doesn't appear on the Tools menu. I use Arduino IDE 2.0.3 on Windows 11.

Thank you!

[image]

[image]

jnegrete2005, 2022-12-31:
I've got the same issue

Jhoff15, 2023-01-09:
got the same issue

Jhoff15, 2023-01-09:
Can someone maybe encrypt my error? I think its because 2.0.3 and Windows 11.

Thats the error:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40203593 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd00 end: 3fffffc0 offset: 0190
3ffffe90:  3ffe85c8 3ffee9fc 00000000 40201148
3ffffea0:  00000000 000d000f 00000000 402077c4  
3ffffeb0:  d202a8c0 3fffff10 3ffffed0 40204051
3ffffec0:  3ffe8b2f 3fffff10 3ffee9fc 40201226
3ffffed0:  00000000 000d000f 00000000 40203700
3ffffee0:  3ffefcac 00000023 3ffee9fc 402038b0
3ffffef0:  3ffee8ac 3ffee838 3ffee9fc 3ffee9fc  
3fffff00:  3ffee8ac 3ffee838 00000017 4020151f
3fffff10:  402077c4 d102a8c0 00000001 00000000  
3fffff20:  00000000 3fffff40 00000001 3ffeea88
3fffff30:  3fffdad0 00000000 3ffeea74 402015c1
3fffff40:  402075f8 00000000 00001388 3ffee854  
3fffff50:  00000000 00000000 00000000 00000000  
3fffff60:  007a1200 b2760af9 00000000 4020452f
3fffff70:  00000000 00000009 3ffefc8c 40201450
3fffff80:  00000000 00000000 00000001 40100164  
3fffff90:  3fffdad0 00000000 3ffeea74 3ffeea88
3fffffa0:  3fffdad0 00000000 3ffeea74 40204780  
3fffffb0:  feefeffe feefeffe 3ffe85ec 40100c95
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Jhoff15, 2023-01-09:
@jnegrete2005 @daniel-ungureanu got it working. I had Arduino IDE Version 2.0.3 on Win11 and i experienced the same issue you had.
You have to use a older Version of Arduino IDE.

Get the Legacy Version of Arduino here: https://downloads.arduino.cc/arduino-1.8.19-windows.exe

Now the Tool shows up in Arduino and you can work with it. You dont even have to deinstall the 2.0.3 version.