me-no-dev/EspExceptionDecoder #34: ERROR: xtensa-lx106-elf-gdb.exe not found!

perigalacticon, 2018-03-27:
I installed EspExceptionDecoder to Documents\Arduino\tools\EspExceptionDecoder\tool
which contains only EspExceptionDecoder.jar.  When I try to start the decoder from the menu the output window shows the error:  ERROR: xtensa-lx106-elf-gdb.exe not found!  I am using v1.8.5 on Windows10 64bit.  I have compiled for esp8266 and esp32 many times previously.

metinkun, 2018-03-29:
i have same issue. v1.0.6 was ok. i think there is something wrong in release v1.1.0

anuroopshannu, 2018-03-31:
ERROR: xtensa-lx106-elf-gdb.exe not found!

any work around for this??

leobel96, 2018-03-31:
Same problem here with Arduino IDE 1.8.5 and Windows 10

pauloeduardosp, 2018-04-04:
I think that this isue was soved in this case
[https://github.com/me-no-dev/EspExceptionDecoder/issues/3](url)

hydra, 2018-05-17:
Same issue, Arduino 1.8.5, ESP8266 2.4.1 installed via boards manager, tool installed as per documentation.  Windows 10 build 17134.

I had a look in the ESP8266 library that the board manager installed, and sure enough GDB in not installed, here's what *was* in the gcc folder that ESP8266 2.4.1 contained.

```
C:\Users\Hydra\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin>dir
 Volume in drive C is W10X64
 Volume Serial Number is 767C-8C5B

 Directory of C:\Users\Hydra\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin

18/05/2015  21:56    <DIR>          .
18/05/2015  21:56    <DIR>          ..
19/04/2015  14:00         1,241,889 libiconv-2.dll
19/04/2015  14:00         5,101,699 xtensa-lx106-elf-addr2line.exe
19/04/2015  14:00         5,371,819 xtensa-lx106-elf-ar.exe
19/04/2015  14:00         7,259,659 xtensa-lx106-elf-as.exe
19/04/2015  14:00         2,333,940 xtensa-lx106-elf-c++.exe
19/04/2015  14:00         5,006,273 xtensa-lx106-elf-c++filt.exe
19/04/2015  14:00         2,330,310 xtensa-lx106-elf-cpp.exe
19/04/2015  14:00           159,202 xtensa-lx106-elf-elfedit.exe
19/04/2015  14:00         2,333,940 xtensa-lx106-elf-g++.exe
19/04/2015  14:00         2,305,587 xtensa-lx106-elf-gcc-4.8.2.exe
19/04/2015  14:00           187,501 xtensa-lx106-elf-gcc-ar.exe
19/04/2015  14:00           187,501 xtensa-lx106-elf-gcc-nm.exe
19/04/2015  14:00           187,501 xtensa-lx106-elf-gcc-ranlib.exe
19/04/2015  14:00         2,305,587 xtensa-lx106-elf-gcc.exe
19/04/2015  14:00         1,195,487 xtensa-lx106-elf-gcov.exe
19/04/2015  14:00         6,184,458 xtensa-lx106-elf-gprof.exe
19/04/2015  14:00         6,970,533 xtensa-lx106-elf-ld.bfd.exe
19/04/2015  14:00         6,970,533 xtensa-lx106-elf-ld.exe
19/04/2015  14:00         5,121,486 xtensa-lx106-elf-nm.exe
19/04/2015  14:00         6,078,914 xtensa-lx106-elf-objcopy.exe
19/04/2015  14:00         6,703,873 xtensa-lx106-elf-objdump.exe
19/04/2015  14:00         5,371,819 xtensa-lx106-elf-ranlib.exe
19/04/2015  14:00         1,052,900 xtensa-lx106-elf-readelf.exe
19/04/2015  14:00         5,158,022 xtensa-lx106-elf-size.exe
19/04/2015  14:00         5,081,525 xtensa-lx106-elf-strings.exe
19/04/2015  14:00         6,078,914 xtensa-lx106-elf-strip.exe
              26 File(s)     98,280,872 bytes
               2 Dir(s)   8,114,229,248 bytes free

```

hydra, 2018-05-17:
This is probably related too https://github.com/CHERTS/esp8266-devkit/issues/27

tarzan115, 2018-07-19:
here is solution for me https://github.com/me-no-dev/EspExceptionDecoder/issues/36#issuecomment-406144393

uched41, 2018-07-28:
This was the folder that worked for me:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\bin

try this.

RobotGarage, 2019-02-21:
This solution worked for me: https://github.com/me-no-dev/EspExceptionDecoder/issues/36#issuecomment-465846882

BisKitGo, 2019-03-23:
Thanks RobotGarage and tarzan115.  #36 (comment) worked for me also

HamzaHajeir, 2019-04-15:
Hello guys, 

What worked well with me is here : 
https://forum.arduino.cc/index.php?topic=540292.msg3705349#msg3705349