me-no-dev/EspExceptionDecoder #36: ERROR: xtensa-lx106-elf-gdb.exe not found!

MSKeith, 2018-04-23:
I just installed the stack trace tool, and I am getting this error message.  I have also searched for the tool chain...and found it can be installed from here:
gnutoolchains.com/esp8266/
but after that installation and re starting the Arduino IDE... I am still not getting the stack trace decoder to start.

Thoughts?

Keith

haxinglong, 2018-04-30:
I  got it too,but I fixed it up.
Try copy xtensa-lx106-elf-gdb.exe to "C:\Users\\***\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin".
It works.
Arduino IDE ver :1.8.0
OS: win7 X64

Traste, 2018-05-02:
@haxinglong 
I would have tried that, but i can not seem to find it in the folder it is supposed to be:
C:\Users\Drufus\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin

beikeland, 2018-05-13:
Duplicate, and according to other issues:

"Since this problem is recent, using the previous release should temporarily do it.
https://github.com/me-no-dev/EspExceptionDecoder/tree/1.0.6"

Worked for me

BoOM26, 2018-05-25:
https://forum.arduino.cc/index.php?topic=540292.0

This helps me...

Tvde1, 2018-06-02:
Getting the same error

randm2001, 2018-07-14:
I download EspExceptionDecoder-1.0.6.zip from 
https://github.com/me-no-dev/EspExceptionDecoder/releases
and there isn't any problem with this release

tarzan115, 2018-07-19:
download toolchain from [this](http://gnutoolchains.com/esp8266/)
and copy file `xtensa-lx106-elf-gdb.exe` to `tools/xtensa-lx106-elf`
it will run perfectly.

ofekp, 2018-07-21:
@tarzan115 for reference, if anybody need this in the future, this should be the folder in which to place the exe file:
`C:\Users\{user_id}\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin`
reference from here `https://forum.arduino.cc/index.php?topic=540292.0`

diraniyoussef, 2018-08-14:
@Tarzan I always had hard times choosing among NonOS SDK and IoT SDK 2.0.0. I'm using NodeMCU Lolin.
What do you suggest?

diraniyoussef, 2018-08-14:
Just to help out the ones (like me) that didn't know where to find xtensa-lx106-elf-gdb.exe after installing the "Prebuilt GNU toolchain for esp8266" linked by tarzan115, it is here "C:\SysGCC\esp8266\bin"

Smeaj, 2018-09-20:
Has anyone else found they need to copy this file again after restarting the IDE?

YuMERA, 2018-11-16:
> @tarzan115 for future reference, if anybody encouters this, the folder to which you need to copy the `xtensa-lx106-elf-gdb.exe` file is (**and not** `arduino-1.8.5\tools\xtensa-lx106-elf`):
> `C:\Users\{user_id}\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin`
> reference from here `https://forum.arduino.cc/index.php?topic=540292.0`
> 
> Also, IDE 1.6.8 did not work with this and I had to update to 1.8.5 to make use of it.
On Win10 and IDE 1.8.5 folder when must copy file `xtensa-lx106-elf-gdb.exe`  is
`C:\Users\{user_id}\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin`

apanasara, 2019-02-08:
tried for win 10 Arduino IDE 1.8.8 for all above suggestions, but failed with error that 

> ERROR: xtensa-lx106-elf-gdb.exe not found!
** @Arduinon Data Folder**
[image]

**@Arduino15 Folder**
[image]

RobotGarage, 2019-02-21:
I was having the same issue in Arduino IDE 1.8.8 (Windows 10):
- ERROR: xtensa-lx106-elf-gdb.exe not found!

This worked for me:
- In Arduino IDE, un-install ESP8266 from >Tools >Boards >Board Manager
- In File Manager, remove "esp8266" folder from here: C:\Users\<UserName>\AppData\Local\Arduino15\packages
- In Arduino IDE, re-install ESP8266 to >Tools >Boards >Board Manager

AndreiKorchunov, 2023-08-03:
Arduino IDE 1.8.19
Boards:
AVR 1.8.3
Esp32 2.0.11
Esp8266 3.1.2
Esp exception decoder 2.0.3

When I try to run the esp exception decoder, I get the following error:

```
gdbPath: C:\Users\kors\Documents\ArduinoData\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823/bin/xtensa-lx106-elf-gdb.exe

ERROR: xtensa-lx106-elf-gdb.exe not found!
```
It can be seen that the problem is that the search for xtensa-lx106-elf-gdb.exe for some reason is carried out along the path C:\Users\kors\Documents\ArduinoData\packages\esp32, although the file itself is located in c:\Users\kors \Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\bin\xtensa-lx106-elf-gdb.exe

Esp exception decoder is installed as recommended by the developer:
c:\Arduino\tools\EspExceptionDecoder\tool\EspExceptionDecoder.jar

The suggested solution above didn't help.
Just in case, I reinstalled c:\Users\kors\Documents\ArduinoData\packages\esp32. Did not help
How to fix?