me-no-dev/EspExceptionDecoder #37: Decode failed

dbischoff777, 2018-05-27:
Greetings!

Sadly i get the error "Decode failed". Arduino  IDE is 1.85 and i'm using the latest Decoder release.

>>>stack>>>
3ffffe70:  3fff0a30 00000000 00000000 4021659b  
3ffffe80:  4010500c 025f5af8 3fff19a4 40212a9f  
3ffffe90:  3ffef470 00000244 00000244 fffffffb  
3ffffea0:  3fff0914 00000000 3fff0910 4020f01d  
3ffffeb0:  40000f68 3ffe933c 3fff0910 40212590  
3ffffec0:  00000000 00000007 3fff0d54 4021201d  
3ffffed0:  3ffe933c 3ffe933c 3fff0910 402125be  
3ffffee0:  00000000 3fff0910 3fff0d54 40214ae0  
3ffffef0:  0100a8c0 3fffdcb0 3ffee918 00000030  
3fffff00:  00000000 3fff0910 3fff0d54 40214b33  
3fffff10:  40210450 3fff056c 3fff0a34 40100690  
3fffff20:  3ffee8f0 3ffe9444 00000030 3fff0568  
3fffff30:  40210450 3fff056c 3ffe92d0 4021045e  
3fffff40:  40104d7e 0243e6ff 00000001 4021051c  
3fffff50:  3ffe92d0 3ffe9444 0243e6ff 00000000  
3fffff60:  4022eb14 3ffee8c8 3ffee8f0 60000600  
3fffff70:  08d6a0e3 3ffee8f0 3ffee8c8 4022eb21  
3fffff80:  4022eb66 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee900 40000f65 3fffdab0 00000000  
3fffffa0:  40000f49 3ffe96f0 3fffdab0 40000f49  
<<<stack<<<

jhmluna, 2018-07-07:
Also with me.
Arduino IDE release 1.8.5 and Windows 10 64 bit does not work.
Gives me Decode Failure error.

madmartin, 2018-07-20:
I never had a successful exception decode with release 1.1.0 on arduino 1.8.5 on linux-64bit.

BTW Release 1.0.6 DOES decode my exceptions, but points to meaningless filenames and line numbers.

What kind of information can I deliver to help find the problem?

jackkum, 2018-07-29:
Downgrade to 1.0.6 fix this issue to me,

madmartin, 2018-07-29:
@jackkum Are you sure? Does 1.0.6 decode your ESP's exceptions in a way that helps you? 
My experience is that 1.0.6 decodes linenumbers that do not exist in the file it is talking about, allover a bunch of useless messages. This should be a known issue with 1.0.6.

jhmluna, 2018-07-30:
I realized that I was using the Arduino IDE installed by Windows Store. After uninstalling and using the installer provided by the Arduino website, put the xtensa-lx106-elf-gdb.exe file in the core package folder of ESP8266, started working.

jackkum, 2018-07-30:
@madmartin Yes. First I downloaded the latest version, then I replaced the file, put the xtensa-lx106-elf-gdb.exe file and I got an error `Decode faild` all the time. Then I downloaded the release 1.0.6, unpacked and replaced all the files and after that I got the human stack.

jhmluna, 2018-08-09:
@madmartin 

1) download ESP8266 Exception Decoder and install according to website guild.
https://github.com/me-no-dev/EspExceptionDecoder

2) download ESP8266 SDK that includes "xtensa-lx106-elf-gdb.exe" file
https://bintray.com/kireevco/generic/download_file?file_path=esp-alt-sdk-v1.5.0.258-windows-x86.zip

3)  unzip esp-alt-sdk-v1.5.0.258-windows-x86.zip file and copy "xtensa-lx106-elf-gdb.exe" to following location
C:\Users\{user_id}\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin

My reference: https://forum.arduino.cc/index.php?topic=540292.0

OneGneissGuy, 2020-07-12:
Release 1.0.6 on Arduino 1.8.10 on Ubuntu 18.04 decodes the stack trace  with ?/?? at line numbers

jbasiago, 2020-07-24:
I'm getting Decode Failed on Mac High Sierra with Release 1.0.6 and 1.1.0 on Arduino 1.8.12 using a NodeMCU1.0 device.  Does this tool support applications that are divided into multiple files.  I have a .h and .cpp file defined for each of 8 classes that I've developed for the application.  It all compiles and runs fine, but I'm getting an exception under certain conditions.  I'm getting no help with this tool.  Any ideas?

Thanks,