me-no-dev/EspExceptionDecoder #38: No output after pasting stack trace (1.1.0)

thekurtovic, 2018-06-13:
I previously had 1.0.6, then updated to the latest version after realizing the update addressed the inaccurate line issue, but I haven't had any success with the updated tool.

I originally was getting the error below, but resolved that after downloading the file and pasting it in the appropriate directory.
ERROR: xtensa-lx106-elf-gdb.exe not found!
https://github.com/me-no-dev/EspExceptionDecoder/issues/36

Although I can open the tool, the problem I have now is that nothing happens after pasting in a stack trace. 

Has anyone else experienced this? I have Arduino 1.8.5, and Windows 7 64 bit.

randm2001, 2018-07-14:
I have this problem too,
Arduino 1.8.5,Win 10 64bit

tarzan115, 2018-07-19:
here is solution https://github.com/me-no-dev/EspExceptionDecoder/issues/36#issuecomment-406144393

thekurtovic, 2018-07-22:
I had previously tried what is shown in that thread, but it did not seem to change anything for me. I tried again as a sanity check, still no dice. The tool opens, it just doesn't decode the stack traces I enter. 
I am copying the file to the correct location, so that's not the issue for me. 
I will reproduce the same steps on a Windows 10 system when I have the opportunity, and share my results.

zainuvk, 2018-07-27:
I had this problem with new version.After pasting the stack trace decoder wasn't decoding the stack trace.Changed the version (https://github.com/me-no-dev/EspExceptionDecoder/releases) to 1.0.6 and then copied EspExceptionDecoder.jar to C:\Users\userid\Documents\Arduino\Tools\EspExceptionDecoder\tool. It worked and decoder was showing the point of exception.Hope this workaround helps.

Regards,
Zainu

coleg012, 2019-05-13:
Thanks, @zainuvk. Downgrading to 1.0.6 helped me with this issue. Although line numbers shows '?' it's still usable to narrow down to function name at fault.

amansinghaljpr, 2021-05-18:
I am facing the same issue with the latest version

Sarapsis, 2021-05-30:
> 
> 
> I am facing the same issue with the latest version

Same Here

MarcelBaas, 2022-11-10:
Running 2.02. Same issue :(

isadorabit, 2023-04-25:
Hi!
The same with me. But a tried somenthing silly e did work!
Before a was pasting just the addresses like this:  0x4008f560 : 0x3ffd0240 0x4008f7dd : 0x3ffd0260 ........
Then a pasted the part " Backtrace: " together and then it worked
Like this: 
Backtrace: 0x4008f560:0x3ffd0240 0x4008f7dd:0x3ffd0260 0x401c52e7:0x3ffd0280 0x401c532e:0x3ffd02a0 0x401c4a07:0x3ffd02c0 0x401c4d3e:0x3ffd02e0 0x401c4aa5:0x3ffd0300 0x400d13b1:0x3ffd0320 0x400d2029:0x3ffd0340 0x400d2806:0x3ffd0360 0x401adc19:0x3ffd0390 0x400f3ba5:0x3ffd03b0 0x400f3a94:0x3ffd03e0 0x400ef746:0x3ffd0400 0x400ef8e9:0x3ffd0540 0x4010c8e1:0x3ffd0560 0x4010e46c:0x3ffd05a0 0x40090892:0x3ffd05d0

closed 2025-01-03 by thekurtovic