me-no-dev/EspExceptionDecoder #41: EspExceptionDecoder-1.1.0.zip - Invalid or corrupt jarfile

gojimmypi, 2018-07-13:
tried to take the jarfile for a test drive. Downloaded, confirmed "unblocked" file, yet get this error message when opening directly (I don't usually use the Arduino IDE)
[image]

fetched from here:

https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/1.1.0

when placed in tool directory:

[image]

I can compile:

[image]

but when I click on Tools - ESP Exception Decoder, I see this error:

[image]

and for the `ERROR: xtensa-lx106-elf-gdb.exe not found!`, sure enough there's not one to be found in my Documents\Arduino\hardware\esp8266com  (or any subdirectories). 

I'm using the git version of the ESP8266 library. (gcc version 4.8.2) 
`SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:94e9704`

So I'm not sure if this is the same or different issue as https://github.com/me-no-dev/EspExceptionDecoder/issues/34 (does the corrupt jar file cause the trigger of gdb, or does the missing gdb make the jar file look corrupt?)

wegunterjr, 2020-07-17:
same issue here...  any updates?

birdoffire1549, 2023-08-31:
Ditto...
Wow, 3 years later than first reported and no responses. 
I guess this project has been abandoned.