me-no-dev/EspExceptionDecoder #42: Decode Failed

vangalvin, 2018-07-17:
Installed everything as per the instructions
Compiled and uploaded the sketch.
Sketch did a stack dump

`>>>stack>>>
3fffeea0:  00000000 000007c6 00000000 3fffee34  
3fffeeb0:  00000000 3fffdad0 3ffef630 00000030  
3fffeec0:  3fffeee0 3fffeee0 00000004 402363c2  
3fffeed0:  00000000 00000008 00000018 00000000  
3fffeee0:  3ffe9504 3ffe9520 000007c6 00000001  
3fffeef0:  00000018 0000000c 00000000 3ffefa3c  
3fffef00:  3fffdc80 3fff1724 3fff2724 3fff1794  
3fffef10:  00000000 3ffedcf8 00000000 00000000  
3fffef20:  402361e0 3ffedcf8 3ffeec60 4902f545  
3fffef30:  40238e29 3ffedcf8 3ffeec60 4902f545  
3fffef40:  40238e6e 3fffdab0 00000000 3fffdcb0  
3fffef50:  3ffeec70 3fffdad0 3ffef630 40210eef  
3fffef60:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffef70:  40000e19 40001878 00000002 00000000  
3fffef80:  3fffff10 ffffffff ffffffff 40104424  
3fffef90:  4010442a 00000002 00000000 0e31c8f1  
3fffefa0:  4010000d 8ae1a40f e577d16c 8249d4fc  
3fffefb0:  40100774 3fffef4c 40100721 3fffff18  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  feefeffe feefeffe feefeffe feefeffe  
3ffff150:  feefeffe feefeffe feefeffe feefeffe  
3ffff160:  feefeffe feefeffe feefeffe feefeffe  
3ffff170:  feefeffe feefeffe feefeffe feefeffe  
3ffff180:  feefeffe feefeffe feefeffe feefeffe  
3ffff190:  feefeffe feefeffe feefeffe feefeffe  
3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff200:  feefeffe feefeffe feefeffe feefeffe  
3ffff210:  feefeffe feefeffe feefeffe feefeffe  
3ffff220:  feefeffe feefeffe feefeffe feefeffe  
3ffff230:  feefeffe feefeffe feefeffe feefeffe  
3ffff240:  feefeffe feefeffe feefeffe feefeffe  
3ffff250:  feefeffe feefeffe feefeffe feefeffe  
3ffff260:  feefeffe feefeffe feefeffe feefeffe  
3ffff270:  feefeffe feefeffe feefeffe feefeffe  
3ffff280:  feefeffe feefeffe feefeffe feefeffe  
3ffff290:  feefeffe feefeffe feefeffe feefeffe  
3ffff2a0:  feefeffe feefeffe feefeffe feefeffe  
[... 189 lines cut ...]
3ffffe80:  00000001 3fffdad0 3ffef630 00000030  
3ffffe90:  00000000 3fffdad0 3ffef630 00000030  
3ffffea0:  00000000 00000000 00000000 fffffffe  
3ffffeb0:  ffffffff 3fffc6fc 00000001 3ffef628  
3ffffec0:  00000000 3fffdad0 3ffef630 00000030  
3ffffed0:  40210fac 40100734 40210fa0 00000000  
3ffffee0:  00000000 4bc6a7f0 a8f5c28f 00000000  
3ffffef0:  00000000 00000000 4bc6a7f0 00000000  
3fffff00:  40104d40 163a4d14 3ffef684 00000000  
3fffff10:  3ffeec60 3ffef684 3ffe862c 3ffef684  
3fffff20:  3fffdad0 3ffef630 40210f34 3fffefb0  
3fffff30:  3ffef684 3fffdad0 00000012 402018eb  
3fffff40:  3fffdad0 00000000 3ffef248 4020a4d5  
3fffff50:  00000500 0000004f 48e4c5f5 00000001  
3fffff60:  00000044 3fff1bc4 3ffef3c0 4021226c  
3fffff70:  00000002 00000000 3ffef3c0 3ffef630  
3fffff80:  3fffdad0 00000000 3ffef248 4020a6b7  
3fffff90:  00000000 00000000 3ffef248 402083e7  
3fffffa0:  3fffdad0 00000000 3ffef244 40208edb  
<<<stack<<<

Cut and Pasted the above to the `Exception Decoder
and exception decoder just outputs "Decode Failed" noting else.

Any ideas what I am missing?

botletics, 2018-07-26:
Yup, same, I get "Decode failed" as well.

thomastech, 2018-08-10:
Decode failed. 
Arduino IDE 1.8.5, ESP Core 2.4.1. Win10 64 bit pro.
I tried all the posted remedies I found, including downgrading to v1.0.6. This issue seems like it has affected a lot of installations.

Tsaukpaetra, 2018-09-09:
Yeah, at least to provide a more useful error message like "Decode failed: Unable to open blargh.bin" or whatever.

Oh well...

GeorgeTsolov, 2019-05-12:
Pl
[image]
ease HELP

GeorgeTsolov, 2019-05-12:
IDE versoin 1.8.9.
NodeMCU
-----------------------------
Exception (29):
epc1=0x40218588 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe10:  00000004 3fff7734 3ffeda58 40235505  
3ffffe20:  40101be1 3ffeda58 3ffeb554 00000018  
3ffffe30:  0000000f d7f0eca5 3ffee4c8 3fff7734  
3ffffe40:  3fff10fc 00000000 3fff10f8 40218639  
3ffffe50:  4021b9a6 3ffe9714 3fff10f8 4021bba8  
3ffffe60:  00000000 3ffffe90 3ffea8a0 00000001  
3ffffe70:  3ffe9714 3ffe9714 3fff10f8 4021bbd6  
3ffffe80:  3fff1cf4 3fff10f8 3fff11b0 4021ddff  
3ffffe90:  c0a80101 40101f8c 3fffc200 00000022  
3ffffea0:  00000000 000000f9 3fff776e 3fff11d6  
3ffffeb0:  000000f5 3fff11b0 000000f0 00000004  
3ffffec0:  00000004 00000004 00000000 3fff7734  
3ffffed0:  00000224 3fff7f18 0000007f 3fff7734  
3ffffee0:  3fff11e0 3fff7766 3fff1b84 3fff7734  
3ffffef0:  3fff11e0 3fff7766 3fff1b84 4021be68  
3fffff00:  00000001 00000043 3fff10f8 00000000  
3fffff10:  00000000 0000024e 3fffff80 3fff7752  
3fffff20:  3fff10f8 3fff7734 3fff11e0 4021f22c  
3fffff30:  00000014 00000001 0000007f 3fff10f8  
3fffff40:  3fffdc80 3fff1ac4 3fff7f1c 3fff2c5c  
3fffff50:  00000008 3fff10f8 3fff7734 4021899d  
3fffff60:  3fffdc80 3fff1ac4 3fff7f1c 4010453c  
3fffff70:  402419de 3fff1ac4 3fff7f1c 402419f0  
3fffff80:  3fff7744 3fff7734 00000000 3fffdcb0  
3fffff90:  4023c8b7 00000000 3fff7f1c 40243957  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

botletics, 2019-05-13:
The ESP8266 decoder tool issue is ESP-related, and is up to the maintainer
of that tool.

As for the exception errors, it could be insufficient power supply or
wiring.