me-no-dev/EspExceptionDecoder #43: Support for Windows 10

yldzmuhammed, 2018-09-09:
Hello,

Does this tool work on Windows 10? 

I have Arduino 1.8.5 and I followed every inst. on the readme. It is not on the tools menu on my computer.

Does this known?What should i do to get it work?

MajicCat, 2018-10-31:
Same here, double checked instructions and tried 3 machines..

quaggy, 2019-03-18:
Same problem. I dopn't think the code is suitable for windows.

smokes2345, 2019-04-18:
Make sure you're using the correct path.  The path to the jar file should look something like this <user>/Documents/Arduino/tools/<tool_name>/tool/<tool_name>.jar.  I'm able to get the tool to load into the IDE and show up in the menu, but when I attempt to paste in a stack trace nothing happens.  I'm afraid the core ESP library may have been updated such that this tool is no longer compatible.  I'd love someone to confirm this or prove me wrong, but recent issue reports have no responses.  I think this thing might be dead.

MkLHX, 2019-09-20:
get the same error, works during few month on Arduino 1.8.9.
Updated today to 1.8.10, nothing on tools. Try apply the workflow installation.
Nothing appear on tools menu.

j-eyking, 2019-09-26:
The same over here: Windows 10 and Arduino 1.8.10, but just upgraded because it didn't appear in 1.8.5 either.

evgesha14, 2020-07-02:
Same issue with Arduino 1.8.13, nothing happens after I paste the stack