me-no-dev/EspExceptionDecoder #47: How to make this work in PlatformIO

starlino, 2019-01-14:
Any help or hints appreciated.

lucasromeiro, 2019-01-15:
I also need

MkLHX, 2019-07-04:
Hi there ! some news about integration in PIO ?

proddy, 2019-07-04:
I used a Python implementation which reads the stack dump from a file. I launch it from Visual Studio Code directly. See https://github.com/proddy/EMS-ESP/blob/master/scripts/analyze_stackdmp.py

starlino, 2019-07-04:
Thanks proddy. 

Q: Is there any way to get /.pio/build/debug/firmware_d1_mini.elf  as a variable so it is not hard-coded in script ??

Anyone game to make an VS Code extension using the Python script ???

proddy, 2019-07-05:
@starlino yes you can get all the pio variables in python so easy to adjust. See the doc https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

Creating a VSC extension isn't that useful. If you want to debug, just copy & paste the stack dump into a file or pipe it into the python script either on a command line or double clicking the file in VSC

qurm, 2020-10-03:
You can use this indirectly to decode a stack trace from PlatformIO, but you do need Arduino installed.  However you don't need to open your PlatformIO project in Arduino, so it is not too hard.
Try this:
1. Install in the Arduino tools as described.
2. In PlatformIO build and run your project and get the stack trace ready in the terminal window.  This generates the .elf file.
3. Open in Arduino menu, Tools, ESP Exception Decoder.
4. You are prompted for a .esp file.   Open from  <project name>/.pio/build/lolin-d32/firmware.elf (or similar)
5. You are prompted for the trace - paste in from the terminal
6. Read the decoded stack results.
[image]

liebman, 2020-10-03:
Since version 4.3 platformio has [monitor_filters](https://docs.platformio.org/en/v4.3.4/core/userguide/device/cmd_monitor.html#cmd-device-monitor-filters) including `esp8266_exception_decoder` and `esp32_exception_decoder`. These will automatically decode the exception.