me-no-dev/EspExceptionDecoder #50: How To: Install and use

eabase, 2019-07-16:
Hey Guys! This is a very cool tool to use and it is very easy. 

Just follow [these installation instructions](https://forum.arduino.cc/index.php?topic=606917.msg4117591#msg4117591) 

When you download into the default Arduino folder, it is owned by Windows Administrator, by default, and will refuse! I suggest to [remove the UAC windows annoyance](https://www.howtogeek.com/howto/4820/how-to-really-completely-disable-uac-on-windows-7/)  before trying to install, unless you do it from an admin CMD shell. 

If you're asked for an ELF file, it simply means you have not yet compiled the sketch, so the program doesn't know where to look for debug traces... Just recompile and open the EspExceptionDecoder. 

:+1:  :1st_place_medal:

Anton-V-K, 2019-09-04:
The tool still complains about missing ELF file, though I've compiled the sketch and uploaded it to the device:
`
Sketch uses 229437 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32124 bytes (39%) of dynamic memory, leaving 49796 bytes for local variables. Maximum is 81920 bytes.
Uploading 233584 bytes from E:\Users\Anton\AppData\Local\Temp\arduino_build_402555/AESTest.ino.bin to flash at 0x00000000
................................................................................ [ 34% ]
................................................................................ [ 69% ]
.....................................................................            [ 100% ]
ERROR: neither AESTest.ino.elf or AESTest.cpp.elf were found!
Did you forget to compile the sketch?
`
ELF-file is in the same dir as BIN-file.

Lomanic, 2019-12-03:
Thanks a lot @eabase, as I use the IDE in "portable" mode I had to create a new folder `portable/sketchbook/tools/EspExceptionDecoder/tool/` and put `EspExceptionDecoder.jar` in it.

I then had the same issue as @Anton-V-K and just recompiled my sketch, following https://github.com/me-no-dev/EspExceptionDecoder/issues/25#issuecomment-320422294.