me-no-dev/EspExceptionDecoder #53: Can this run on windows? I'm on Win 10 Pro and it is asking me for elf lol

Sax-Ted, 2020-04-18:


reivaxy, 2020-04-18:
Yes it runs on Win 10 Pro
Can you provide more context ? Are you using Arduino IDE, or Platformio, for instance?

Sax-Ted, 2020-04-18:
@reivaxy I am using Arduino IDE and it is just asking me if I got some ELF file and idk what just happened haha

Sax-Ted, 2020-04-18:
@reivaxy If u want the screenshot here it is 
[image]

reivaxy, 2020-04-18:
did you compile the sketch ?

Sax-Ted, 2020-04-18:
@reivaxy Oh after compiling it can work?

Sax-Ted, 2020-04-18:
also this isn't the code i am using lol

reivaxy, 2020-04-18:
Well, if you don't compile, you won't have the elf file...
But if you don't compile, you don't run, and then you don't need EspExceptionDecoder

Sax-Ted, 2020-04-18:
AHA gottem, tks man

evgesha14, 2020-07-02:
@Sax-Ted are you able to use this tool on windows (10 i assume)?

Sax-Ted, 2020-07-02:
Uh yea it works when I compile it, sorry!

shivan, 2021-02-13:
Same here, would be great if it would tell something about "please compile first" instead of "select elf" :-)

trullock, 2021-03-03:
How do you have a stack trace if you haven't compiled and run the code first?

alastaira, 2021-03-06:
> 
> 
> How do you have a stack trace if you haven't compiled and run the code first?

In my case, the fact that I had already uploaded code that caused my ESP32 to throw an error stack trace is the thing that led me to Google and find this tool in the first place - so, I have already compiled and uploaded the code on my board, and simply enabling the serial monitor allows me to copy and receive the stack trace - it wasn't clear from the instructions of the description of "Select .elf file" that an additional compilation is required after having installed the tool