me-no-dev/EspExceptionDecoder #57: Running on Windows10

kurtwerbrouck, 2021-02-16:
hello,

i am trying to get the decoder installed and working. 
I tried all what i can but still no succesfull.
example i have a test.ino in the foder c:\test, then i create the folders
tools
EspExceptionDecoder
tool
EspExceptionDecoder.jar (is this also a folder a the file name?)
=> C:\test\tools\EspExceptionDecoder\tool\EspExceptionDecoder.jar.

When i am unzipping the decoder file i place them into the EspExceptionDecoder.jar folder?

If the EspExceptionDecoder.jar is the actual program, where is it located in the zip folder?

Hope you can help

trullock, 2021-03-03:
`tools\EspExceptionDecoder\tool\EspExceptionDecoder.jar` should be under your Arduino folder, probably in `Program Files (x86)`

rpehlm, 2021-03-09:
Windows 10
I have the EspExceptionDecoder.jar file installed in the correct folder.  When I access the tool via the 'Tools' menu, 'ESP Exception Decoder' item, a file browser window is opened looking for an *.elf file.  Makes me think this tool is meant for Linux only!  The documentation does not state Linux only, so I'm not sure.  The only reference to Linux is a comment to the effect that perhaps an updated library is needed on newer distributions.  So, the question is: Will this tool work on Windows 10?

Kurkyxp, 2021-03-11:
Compile (upload not necessary) your sketch and try again

SunBeenMoon, 2021-05-06:
Hi, I'm stuck in same problem. 
@rpehlm @Kurkyxp are you succeed to build this tool in window 10?

pgrawehr, 2021-05-15:
I was just testing it in Windows 10, works as expected. I'm running Arduino IDE 1.8.13 (not the Windows Store Version). I have copied the folder "EspExceptionDecoder" from the Zip file to C:\Users\<Username>\Documents\Arduino\tools. 
When I start it from the Tools menu, it asks me for a stack trace, and I paste the line that starts with "Backtrace" from the serial console into that window. Immediately, I get a nice looking stacktrace that tells me exactly where the code crashed.