me-no-dev/EspExceptionDecoder #60: Need Help :File Extension is JAVA instead jar

joko2020, 2021-06-25:
@me-no-dev hello,

I cannot find the jar file as your description... i put unziped folder as instructed , in Arduino I.D.E 1.8.12 the tool not show
please help

joko2020, 2021-06-25:
ok problem solved  [https://forum.arduino.cc/t/can-not-get-stack-trace-decoder-to-load/582923/2](url) case close

closed 2021-06-25 by joko2020

hondaman900, 2021-07-02:
What was the solution? The link you provide is broken, and I have the same question as you resolved. Thx.

joko2020, 2021-07-24:
> What was the solution? The link you provide is broken, and I have the same question as you resolved. Thx.
Sorry , When I was in that problem , the link is still active 
bellow ill  provide the file
[EspExceptionDecoder.zip](https://github.com/me-no-dev/EspExceptionDecoder/files/6872283/EspExceptionDecoder.zip)