me-no-dev/EspExceptionDecoder #68: Do not decode ESP32 compiled with gcc8_4_0-esp-2021r2

cicciocb, 2021-11-11:
When compiling with latest arduino-ESP32 2.0.1(using the compiler gcc8_4_0-esp-2021r2) an error message is shown "Decode failed".

Using the previous version of the compiler (using the compiler gcc8_4_0-esp-2021r1) works properly.

[image]

cicciocb, 2021-11-11:
Finally the problem comes from xtensa-esp32-elf-gdb.exe that has not been compiled using static libraries so it miss many .dll to work

pilnikov, 2021-11-21:
++

witnessmenow, 2021-11-28:
Is this an ESP32 Core issue? Is there a issue raised over there?

I have an issue on 2.0.1 and I am basically lost without this 😄

cicciocb, 2021-11-28:
@witnessmenow 
It is an issue with the compiler gcc8_4_0-esp-2021r2 that is part of the 2.0.1.
The **"xtensa-esp32s2-elf-gdb.exe"** inside the folder has been compiled to used with external dll that are missing in the folder (not statically compiled like the others .exe).
Simply copy these 2 dll in the folder **packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2\bin** and voilà.
[gdb_dll_missing.zip](https://github.com/me-no-dev/EspExceptionDecoder/files/7613067/gdb_dll_missing.zip)

witnessmenow, 2021-11-28:
Thanks @cicciocb

smuellener, 2022-01-19:
> 

@cicciocb where is this folder "packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\bin" exactely? I am on arduino core 2.0.2 tag and can not find it in the folder ide\hardware\espressif\esp32. Thanks

cicciocb, 2022-01-20:
> > 
> 
> @cicciocb where is this folder "packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\bin" exactely? I am on arduino core 2.0.2 tag and can not find it in the folder ide\hardware\espressif\esp32. Thanks

Hi @smuellener,
the folder is generally located here :  

C:\Users\YOUR_NAME\AppData\Local\Arduino15

smuellener, 2022-01-20:
I have the following folder as custom installation folder for the IDE: C:\Users\Public\arduino\ide\. That resulted in having to copy the files into the following folder:

C:\Users\Public\arduino\ide\hardware\espressif\esp32\tools\xtensa-esp32-elf\bin

Thanks @cicciocb !

pgrawehr, 2022-02-19:
It appears the bug has been fixed on master. With xtensa-esp32-elf-g++ (crosstool-NG esp-2021r2-patch2) 8.4.0 the exception decoder works again (without having to copy the files from the above zip)

me-no-dev, 2022-02-20:
https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/2.0.2 works for all

closed 2022-02-20 by me-no-dev

mcpicoli, 2022-03-03:
Hi,

Just to help people avoiding all of the trouble I had to get this working. TL;DR: _Manual Install is the way_

- Simply downloading the latest release of the EspExceptiodDecoder (2.0.2) and installing _does not solve the problem_.
- Simply re-downloading the ESP32 SDK using the Arduino Board Manager (2.0.2 coincidentally as well) _does not solve the problem_
- Nuking and reinstalling the ESP32 packages ("packages" and "staging/packages" - download cache) _does not solve the problem._
- Installing the 2.0.2 ESP32 Board _for the *first time ever* may solve the problem (*not tested*).
- Deleting the "package_esp32_index.json" file (pointed to at the [installation instructions](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json)) _does not solve the problem_. It just downloads the old versions of the toolchain!

Finally...

- Installing the Arduino ESP32 SDK *manually* (from [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)) and removing the old URL for the board manager *works*. Get ready for a few GB of downloads...

I believe that the board definition files from the official Espressif repository on Github are outdated, using the old toolchain, and the EspExceptionDecoder will _never_ work "out of the box" using the standard, easy procedures of installation, until they're updated to use the newer toolchain.

Asy508, 2023-07-12:
Thanks a lot. I looking this solution.