me-no-dev/EspExceptionDecoder #69: Select ELF?

SteveRMann, 2021-11-19:
"ERROR: elf was not found!"

What is ELF and what do I select?

AcuarioCat, 2021-11-21:
You need to navigate to the folder where your <project>.ino.bin file is written. There you will also find the elf file.

Definition: Elf (Executable and Linking Format) is a common, standard format for executable files, object code, and shared libraries. It is created by avr-gcc as the output of the compilation.

quinman24, 2021-11-30:
what is the directory to the .elf file I need to finger out what is going on with my esp8266 esp-01 board. "You need to navigate to the folder where your .ino.bin file is written." I'm a nob at finding folders where my sketch writes its .ino.bin files go.
[image]

riban-bw, 2021-12-11:
The README does not mention that you may get a pop-up asking for the location of the ELF. The location may be deduced by looking at the last few lines of the compile output. On a Windows machine this might be:
```
"C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-size" -A "C:\\Users\\username\\AppData\\Local\\Temp\\arduino_build_793312/controller.ino.elf"
Sketch uses 396181 bytes (37%) of program storage space. Maximum is 1044464 bytes.
```
In this case the ELF is located in `C:\Users\username\AppData\Local\Temp\arduino_build_793312`

It seems that this is detected after building the project so build, run, debug seems to work but if you haven't built since starting IDE you need to know where the last build put its files.

Note that Arduino IDE cleans the temp files so these may not even be there anymore if you have performed other actions like building a different sketch or restarted IDE.

I recommend this is explained in the README, i.e. that the decoder needs to be run after building the image.

877dev, 2022-01-02:
Same issue here, could not find the ELF file even in `C:\Users\username\AppData\Local\Temp\arduino_build_xxxxxx` . Restarted IDE and then it worked by itself, did not need to locate an ELF file.

Batwam, 2022-05-13:
not super clear where to find it but it can be found with a bit of digging or putting compile with verbose mode. On Linux the elf files is saved in the tmp folder: `/tmp/arduino_build_xxxxxx`

DmitriyVovk, 2023-08-08:
Hello. A have faced the same issue. Because i use VS Code + Visual Micro plug-in the tool requires to compile my code without errors using Arduino IDE. Than will find .elf itself.

mistergreen, 2025-02-02:
I installed the decoder and compile the app, The decoder seems to figured out the elf file directory/location. I'm on OSX so the elf file is hidden in the /var folder.