me-no-dev/EspExceptionDecoder #72: "Arch Not Found for DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

EkremVK, 2022-02-04:
Hi all,

When I click ESP Exception Decoder below the tools menu in Arduino IDE, I get the message in the headline.

I have put EspExceptionDecoder.jar file under C:\Program Files (x86)\Arduino\tools\EspExceptionDecoder\tool on Windows 10 and ~/home/Arduino/tools/EspExceptionDecoder/tool on Ubuntu 18.04 . I get the error on both OSs.

I have tried choosing different boards from the Board menu, none have worked.

I have already looked for a solution online, but found nothing.

Looking at the source code, I found the lines below are executed, but I don't know why.
if(core.contentEquals("esp32")){
      try {
        tarch = BaseNoGui.getBoardPreferences().get("build.tarch");
        if(tarch == null || tarch.contentEquals("")){
          **editor.statusError("Arch Not Found for "+BaseNoGui.getBoardPreferences().get("name"));**
          return;
        }

It would be great if anyone can help.

me-no-dev, 2022-02-04:
In order to support all chips, the proper toolchain needs to be selected. I do not know why this is failing for you... could you try with another board definition? Also, which version of ESP32-Arduino are you running? This will work only on v2.x and up

me-no-dev, 2022-02-04:
maybe I should add a default value for older Arduino versions...

me-no-dev, 2022-02-04:
try this: https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/2.0.1

closed 2022-02-04 by me-no-dev