me-no-dev/EspExceptionDecoder #74: ERROR: null-esp32-elf-gdb.exe not found!

IsaacFroes, 2022-02-07:
I'm having a crash in some random moments in my program. I was hoping to use the decoder to read the backtrace but I'm having the error below after I try to run the decoder.

Arch Not Found for 'ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None'. Defaulting to 'xtensa'
Target Not Found for 'ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None'. Defaulting to 'esp32'
ERROR: null-esp32-elf-gdb.exe not found!

[image]

vicatcu, 2022-02-07:
I was getting the same error using the 1.6.0 package, but the problem was fixed when I updated to the latest 2.2.0

beniroquai, 2022-02-08:
Same here on Mac Os 10.15.7. I am using the latest version from the release tab.

IsaacFroes, 2022-02-08:
I was using the latest release (2.0.1). 
I fix the problem by downgrading to the 1.1.0.
Thanks for the help. 😃

closed 2022-02-08 by IsaacFroes