me-no-dev/EspExceptionDecoder #75: ERROR: null-esp32-elf-gdb.exe not found!

Aviortheking, 2022-02-09:
Following #74, I don't think downgrading is a good idea.

I'm on Windows 11 using EspExceptionDecoder 2.0.1

Error:
```
Arduino: 1.8.16 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 966170 bytes (73%) of program storage space. Maximum is 1310720 bytes.

Global variables use 31956 bytes (9%) of dynamic memory, leaving 295724 bytes for local variables. Maximum is 327680 bytes.

Arch Not Found for 'ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None'. Defaulting to 'xtensa'

Target Not Found for 'ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None'. Defaulting to 'esp32'

ERROR: null-esp32-elf-gdb.exe not found!
```

KaranRajPradhan, 2022-02-12:
I'm facing the same issue:

````
Arch Not Found for 'OLIMEX ESP32-EVB, 80MHz, 115200, Default'. Defaulting to 'xtensa'
Target Not Found for 'OLIMEX ESP32-EVB, 80MHz, 115200, Default'. Defaulting to 'esp32'

ERROR: null-esp32-elf-gdb not found!
````
@Aviortheking were you able to fix it?

Aviortheking, 2022-02-12:
No :(

nfeuer, 2022-02-15:
I'm having the same issue on MacOs

me-no-dev, 2022-02-16:
I'll fix this, but for now use the older exception decoder for Arduino ESP32. Interesting why it fails for 2.0.1

A-R-S-D, 2022-02-19:
Me, too! :(

me-no-dev, 2022-02-20:
https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/2.0.2 should be ok now. missed one declaration... duh

A-R-S-D, 2022-02-20:
Yey, It works now. Thanks.
There is only a warning which I don't think it's important:

> "Arch Not Found for 'DOIT ESP32 DEVKIT V1, 80MHz, 921600, None'. Defaulting to 'xtensa' "

me-no-dev, 2022-02-20:
yes, it just confirms that you are using an older core and it defaulted to xtensa, since that is all we supported before 2.0.0

closed 2022-02-20 by me-no-dev