me-no-dev/EspExceptionDecoder #76: Decode Failed

Grutsch, 2022-03-05:
I am using the latest ExceptionDecoder (2.02) also tried earlier ones but doesn't matter what I do I get Decode Failed (IDE 1.8.19).

[image]

Speedy2k, 2022-03-07:
Same here also! Arduino Core v2.0.2

Speedy2k, 2022-03-07:
2.0.2

drmpf, 2022-03-07:
Same here V2.0.2 of the decoder.  
Looking at the code it seems there is a problem opening/reading the input stream.
You swallow a number of exceptions there.  
Suggest you re-wrap them in RunTimeExceptions and re-throw them and catch the handle (print) them at the higher level for better error messages.

me-no-dev, 2022-03-08:
ok, i guess we are in situation where the latest arduino-esp32 is also 2.0.2 :) Which version arduino-esp32 are you using?

drmpf, 2022-03-08:
I am using Arduino IDE V1.8.19 and ESP32 board support V2.0.2 installed from this url
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
 and the decoder from your releases page (as of 7th March 2022)
p.s. this decoder is invaluable.  I am lost without it :-(

me-no-dev, 2022-03-08:
Except to build a version for you to try, that has more debug enabled, I do not know what to do. I am using the same decoder and it works nicely here (Mac, but should not matter). Latest version should work fine on all chips that are currently supported too (and any future ones, hopefully)

Grutsch, 2022-03-08:
No problem I am willing to help. So just let me know where to pick up this debug version

mcpicoli, 2022-03-08:
Please see the discussion on #68, already closed.

TL;DR: Install the ESP32 board package manually from Espressif's GitHub repo.

The long version: The official board definition package from Espressif for the ESP32 architecture as referenced by @drmpf is still outdated and does not download the latest toolchain ("-patch2" version). If you manually install the ESP32 board package, it should work.

There is no problem with the EspExceptionDecoder (2.0.2) _per se_, but it would be *very* useful to have more detailed error messages than a simple "Decode Failed".

The instructions for the manual install are [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html).

drmpf, 2022-03-09:
I can confirm that adding the .dll's from the zip file mentioned #68 
Fixed my issue and that I can not decode traces from ESP23 V2.0.2 installed via url  
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

Grutsch, 2022-03-09:
For me, the solution described by drmpf is working.
So I will close this issue.
Thanks for the help.

closed 2022-03-09 by Grutsch