me-no-dev/EspExceptionDecoder #79: Decode failed with generated package_xx_index.json

tobozo, 2022-03-19:
Hi and thanks for the awesome plugin!

I get a `Decode failed` error message when pasting exceptions in the decode window although I know the exception can be decoded.

- Platform is Arduino 1.8.19 on Linux amd64
- Package version is 2.0.2
- :warning: Package folder is `~/.arduino15/packages/esp32-2.0.2/` 
- :warning: [package_xx_index.json](https://phpsecu.re/esp32/packages/package_esp32_esp32-2.0.2_index.json) is customised accordingly.
- :warning: platform.txt is also using `name=ESP32 Arduino 2.0.2` instead of `name=ESP32`

[image]

[image]

Note: using a custom package_index.json lets me have several **isolated** package versions in the arduino menu, which is a relief for library development and local testing. 
Apart from the exception decoder, every other tool work fine, and no other side effect from using this custom layout has been observed so far.

[image]

me-no-dev, 2022-03-21:
I have seen this same issue on my end as well. I think it's related to the toolchain, because trying to decode directly also does not produce anything. Previous versions Arduino-ESP32 work ok?

tobozo, 2022-03-21:
I've tested `printf("%s, millis())` crash code on all 2.x.x versions and the behaviour is similar, but I need a better crash code as 1.0.6 compiler prevents such usage :-)

The other java tool I'm using `ESP32 Sketch Data Upload` is not affected and works as expected.

More details on the custom package:

The `name` and `packager` properties have been changed so that the boards manager does not mix paths when searching for tools, libraries, etc.

[image]

Every `platform.txt` uses a modified `name` property, this is to avoid duplicate/nonsensical `ESP32 Arduino` entries in the boards menu. 

[image]

me-no-dev, 2022-03-21:
so you say the same result with 2.0.1 and 2.0.0 and 1.0.6? And you are using the proper toolchains for the versions? As I said, the toolchain is the one not returning anything from parsing the backtrace, it's not the exception decoder that does not rpint it.

@igrr any clues?

tobozo, 2022-03-21:
Do I need to setup a specific environment to recompile this plugin?

I've already played with Arduino IDE Java source in the past so I can eventually do some test/debug.

me-no-dev, 2022-03-21:
Try this:
[EspExceptionDecoder-2.0.2-1-g349d17e.zip](https://github.com/me-no-dev/EspExceptionDecoder/files/8315344/EspExceptionDecoder-2.0.2-1-g349d17e.zip)

tobozo, 2022-03-21:
I modified the plugin to print the dbg command in the textarea, recompiled, and got this when pasting a stack trace:

```
/home/tobozo/.arduino15/packages/esp32-2.0.2/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/
bin/xtensa-esp32-elf-gdb --batch /tmp/arduino_build_958796/crash-esp32.ino.elf -ex set listsize 1 
-ex l *0x4008732e -ex l *0x400e17cb -ex l *0x400e6855 -ex l *0x400e689b -ex l *0x400d14a2 
-ex l *0x400d102d -ex l *0x400d1912 -ex q
```

running that command from a shell returns an obvious error since there's no virtualenv in that shell, but it confirms that the gdb tool is reachable.

`xtensa-esp32-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0`

me-no-dev, 2022-03-22:
BTW the zip I gave you up there outputs more than just the gdb command being executed :) glad you figured it out though. I guess we are waiting on the toolchain team to get this resolved.

tobozo, 2022-03-22:
When I paste this stack trace:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40087331  PS      : 0x00060f30  A0      : 0x800e17ce  A1      : 0x3ffb2360  
A2      : 0x00000384  A3      : 0x00000380  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb26b0  
A10     : 0x3ffc15a8  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060f20  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000384  LBEG    : 0x40087331  LEND    : 0x40087341  LCOUNT  : 0xffffffff  

Backtrace:0x4008732e:0x3ffb23600x400e17cb:0x3ffb2370 0x400e6855:0x3ffb2680 0x400e689b:0x3ffb2710 0x400d14a2:0x3ffb2750 0x400d102d:0x3ffb27f0 0x400d1912:0x3ffb2820 

ELF file SHA256: 0000000000000000
```

I get this output In the Arduino Log window:

```
"/home/tobozo/.arduino15/packages/esp32-2.0.2/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-gdb" "--batch" "/tmp/arduino_build_773672/crash-esp32.ino.elf" "-ex" "set listsize 1" "-ex" "l *0x40087331" "-ex" "q"
"/home/tobozo/.arduino15/packages/esp32-2.0.2/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-gdb" "--batch" "/tmp/arduino_build_773672/crash-esp32.ino.elf" "-ex" "set listsize 1" "-ex" "l *0x00000384" "-ex" "q"
"/home/tobozo/.arduino15/packages/esp32-2.0.2/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-gdb" "--batch" "/tmp/arduino_build_773672/crash-esp32.ino.elf" "-ex" "set listsize 1" "-ex" "l *0x4008732e" "-ex" "l *0x400e17cb" "-ex" "l *0x400e6855" "-ex" "l *0x400e689b" "-ex" "l *0x400d14a2" "-ex" "l *0x400d102d" "-ex" "l *0x400d1912" "-ex" "q"
Decode Failed
```

And I get this the plugin window:

[image]

I've tried this command with libpython2.7 package from different architectures (armhf, arm64) to no luck: I still get the `cannot open shared object file: No such file or directory` or `wrong ELF class: ELFCLASS64` with libpython2.7.so.1.0, so.

WebDust21, 2022-05-16:
Had a very accomplished Linux friend help me work through this issue...
first was copying the commands out of the Arduino log window, and trying them in the Terminal.  The following error resulted:
`........xtensa-esp32-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory`

At least in my case (Ubuntu 22.04 LTS), I only needed to install "libpython2.7"--and further attempts to create a symlink resulted in a "file exists" error.  Ran the exception decoder--and it worked perfectly right away.

`sudo apt-get install libpython2.7`

@tobozo I can decode your stacktrace as well--though of course it's completely useless when based off my project's ELF file ;-)

tobozo, 2022-05-17:
@WebDust21 what's the architecture on your Ubuntu 22.04? 
Doing this on my Ubuntu 20.04 for x64 results in `wrong ELF class: ELFCLASS64` error message.

> libpython2.7 is already the newest version (2.7.18-1~20.04.1).

WebDust21, 2022-05-17:
I'm also on 64-bit (x86_64).

A terminal run of "ldconfig -v | grep libpython" returns the following (after a bunch of "given more than once" notices):
```
libpython2.7.so.1.0 -> libpython2.7.so.1.0
libpython3.10.so.1.0 -> libpython3.10.so.1.0
```

I'm not necessarily a Linux guru, but maybe something will jump out ;-)

tobozo, 2022-05-17:
not that different, but those are x64 versions and expected to be there anyway

        libpython3.9.so.1.0 -> libpython3.9.so.1.0
        libpython3.8.so.1.0 -> libpython3.8.so.1.0
        libpython2.7.so.1.0 -> libpython2.7.so.1.0

[image]

What my system is really missing is `/usr/lib/[some other architecture]/libpython2.7.so.1.0`. However it's already partially rotting with three different versions of python already, and it feels very much like this particular dependency come as part of the espressif sdk, and not be installed by hacking my own OS.

closed 2023-04-08 by tobozo