me-no-dev/EspExceptionDecoder #80: EspExceptionDecoder 2.0.2 for ESP32C3 shows nothing.

janczeresnia, 2022-04-12:
I paste the below memory stack from ESP32C3 to EspExceptionDecoder and nothing is decoded in the Arduino 1.8.19 decoder window. 
What am I doing wrong?
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
MEPC    : 0x40384f30  RA      : 0x4038665c  SP      : 0x3fc8d0b0  GP      : 0x3fc8b600  
TP      : 0x3fc89bcc  T0      : 0x37363534  T1      : 0x42003d94  T2      : 0x33323130  
S0/FP   : 0xffffffff  S1      : 0x00000000  A0      : 0x3fc93c70  A1      : 0x3fc93b0c  
A2      : 0x00000018  A3      : 0x3fc93b0c  A4      : 0x00000018  A5      : 0x3fc93b0c  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0xffffffff  S3      : 0x3fc93c70  
S4      : 0x3fc8d000  S5      : 0x000000d1  S6      : 0x000000aa  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000018  MTVAL   : 0xfec77de3  
MHARTID : 0x00000000  

Stack memory:
3fc8d0b0: 0x3fc8d13c 0x00000020 0x3fc8d0e8 0x3fc93c70 0xffffffff 0x00000000 0x3fc93c4c 0x40385abc
3fc8d0d0: 0x3fc8d130 0x00000002 0x00000001 0xffffffff 0x00000000 0x00000000 0x00000000 0x00000e54
3fc8d0f0: 0x000000aa 0x3fc8d16c 0x3fc8d16c 0x00000015 0x00000001 0x3fc8d17c 0x3fc8afdc 0x42005b0c
3fc8d110: 0x000000aa 0x3fc8d16c 0x3fc8d000 0x0000000e 0x000000ff 0x3fc8c7ec 0x0000000e 0x42003da2
3fc8d130: 0x00006161 0x3fc8c000 0x3fc8c000 0x42004446 0x000000ff 0x3fc8c000 0x3fc8c000 0x4200043e
3fc8d150: 0x00000000 0x3fc8d7fc 0x00000000 0x3fc8fca8 0x0000001f 0x00000015 0x0e006e61 0x00006161
3fc8d170: 0x00000000 0x00000000 0x82000000 0x20766352 0x6d6f7266 0x7830203a 0x8e006161 0xfa32eb3e
3fc8d190: 0x00000000 0x00000000 0x00000000 0x00000001 0x00000000 0x3fc8aff4 0x00000004 0x4038025c
3fc8d1b0: 0x00001881 0x80000003 0x3fc8c69c 0x00000001 0x00001881 0x8000000b 0x3fc8afdc 0x4038019c
3fc8d1d0: 0x00000001 0x3fc8d1d8 0xffffffff 0x3fc90c70 0x3fc90c70 0x00000000 0x3fc8d1ec 0xffffffff
3fc8d1f0: 0x3fc8d1ec 0x3fc8d1ec 0x00000000 0x3fc8d200 0xffffffff 0x3fc8d200 0x3fc8d200 0x00000000
3fc8d210: 0x3fc8d214 0xffffffff 0x3fc8d214 0x3fc8d214 0x00000000 0x3fc8d228 0xffffffff 0x3fc8d228
3fc8d230: 0x3fc8d228 0x00000000 0x3fc8d23c 0xffffffff 0x3fc8d23c 0x3fc8d23c 0x00000000 0x3fc8d250
3fc8d250: 0xffffffff 0x3fc8d250 0x3fc8d250 0x00000000 0x3fc8d264 0xffffffff 0x3fc8d264 0x3fc8d264
3fc8d270: 0x00000000 0x3fc8d278 0xffffffff 0x3fc8d278 0x3fc8d278 0x00000000 0x3fc8d28c 0xffffffff
3fc8d290: 0x3fc8d28c 0x3fc8d28c 0x00000000 0x3fc8d2a0 0xffffffff 0x3fc8d2a0 0x3fc8d2a0 0x00000000
3fc8d2b0: 0x3fc8d2b4 0xffffffff 0x3fc8d2b4 0x3fc8d2b4 0x00000000 0x3fc8d2c8 0xffffffff 0x3fc8d2c8
3fc8d2d0: 0x3fc8d2c8 0x00000000 0x3fc8d2dc 0xffffffff 0x3fc8d2dc 0x3fc8d2dc 0x00000000 0x3fc8d2f0
3fc8d2f0: 0xffffffff 0x3fc8d2f0 0x3fc8d2f0 0x00000000 0x3fc8d304 0xffffffff 0x3fc8d304 0x3fc8d304
3fc8d310: 0x00000000 0x3fc8d318 0xffffffff 0x3fc8d318 0x3fc8d318 0x00000000 0x3fc8d32c 0xffffffff
3fc8d330: 0x3fc8d32c 0x3fc8d32c 0x00000000 0x3fc8d340 0xffffffff 0x3fc8d340 0x3fc8d340 0x00000000
3fc8d350: 0x3fc8d354 0xffffffff 0x3fc8d354 0x3fc8d354 0x00000000 0x3fc8d368 0xffffffff 0x3fc8d368
3fc8d370: 0x3fc8d368 0x00000000 0x3fc8d37c 0xffffffff 0x3fc8d37c 0x3fc8d37c 0x00000000 0x3fc8d390
3fc8d390: 0xffffffff 0x3fc8d390 0x3fc8d390 0x00000000 0x3fc8d3a4 0xffffffff 0x3fc8d3a4 0x3fc8d3a4
3fc8d3b0: 0x00000000 0x3fc8d3b8 0xffffffff 0x3fc8d3b8 0x3fc8d3b8 0x00000000 0x3fc8d3cc 0xffffffff
3fc8d3d0: 0x3fc8d3cc 0x3fc8d3cc 0x00000000 0x3fc8d3e0 0xffffffff 0x3fc8d3e0 0x3fc8d3e0 0x00000000
3fc8d3f0: 0x3fc8d3f4 0xffffffff 0x3fc8d3f4 0x3fc8d3f4 0x00000002 0x3fc8d408 0xffffffff 0x3fc8f898
3fc8d410: 0x3fc93af8 0x00000000 0x3fc8d41c 0xffffffff 0x3fc8d41c 0x3fc8d41c 0x00000000 0x3fc8d428
3fc8d430: 0x00000000 0x00000000 0x00000000 0x3fc8d440 0xffffffff 0x3fc8d440 0x3fc8d440 0x00000000
3fc8d450: 0x3fc8d454 0xffffffff 0x3fc8d454 0x3fc8d454 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc8d470: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc8d47c 0x00000000 0x00000000 0x00000000
3fc8d490: 0x3fc8d494 0xffffffff 0x3fc8d494 0x3fc8d494 0x00000000 0x3fc8d4a8 0xffffffff 0x3fc8d4a8

```

kaizoku-oh, 2022-04-19:
The same happened with me, it would be helpful if we know if there is a support for esp32c3 (riscv) exceptions decoding

beniroquai, 2023-05-16:
Any update here? :)

vlastahajek, 2023-06-12:
The workaround is changing the stack header, `Stack memory:` -> `>>>stack>>>`

vlastahajek, 2023-06-12:
When you are copy-pasting stack from the device console, change it in the ExceptionDecoder window:|
[image]