me-no-dev/EspExceptionDecoder #84: unclear instruction

Patsjemoe, 2022-11-06:
Hi,
I am confused with the install instructions (maybe I am not sufficiently experienced ?) 
Install instruction says : 
1) **In your Arduino sketchbook directory**, create tools directory if it doesn't exist yet.
2) Unpack the tool into tools directory (the path will look like <home_dir>/Arduino/tools/EspExceptionDecoder/tool/EspExceptionDecoder.jar).
When I read this, it seems to do 2 different things.
In 1) , I would expect arduino/hardware/espressif/esp32/**sketches**/TOOLS
in 2) tools is created in arduino directory, not in sketches.
Please clarify
regards
Ludo

AlexP210, 2022-11-06:
I just installed that part successfully by creating the tools folder in the default location that Arduino IDE saves sketches to. For me it's C:\Users\<username>\Documents\Arduino (I'm on Windows). I think you should be able to find the equivalent folder on your system by creating a blank sketch, click save, and see which folder Arduino IDE opens in the "Save sketch folder as" window.

Patsjemoe, 2022-11-06:
Hi Alex,
Thank you for your feedback.
Jut found out that it does not work with arduino 2.0....installed
1.18....and there it shows up..... But could not use it because get too
many compiling errors...
Is there no version that supports 2.0?
Regards
Ludo