me-no-dev/EspExceptionDecoder #87: Standalone Version Without Arduino IDE?

wachidsusilo, 2022-12-13:
Hi, is it possible to use this plugin as standalone app (without Arduino IDE) ?

bobemoe, 2023-02-23:
I just found this https://github.com/luffykesh/EspExceptionDecoder-CLI and it seems to be working :)

wachidsusilo, 2023-08-06:
Almost forgot about this issue. I have made a windows app for this.
Here is how the app looks like:

[image]

Elektrik1, 2023-10-21:
> Almost forgot about this issue. I have made a windows app for this. Here is how the app looks like:
Hi. Can we download it somewhere?

hitecSmartHome, 2024-09-18:
> > Almost forgot about this issue. I have made a windows app for this. Here is how the app looks like:
> 
> Hi @wachidsusilo. Can we download it somewhere?

Would be good to know!