me-no-dev/EspExceptionDecoder #88: Error: Invalid or corrupt jarfile

alorbach, 2023-06-13:
Am I the only one with that error regardless of the EspExceptionDecoder Version?
[image]

System is latest Windows 11 with:

> openjdk version "11.0.12" 2021-07-20
> OpenJDK Runtime Environment Microsoft-25199 (build 11.0.12+7)
> OpenJDK 64-Bit Server VM Microsoft-25199 (build 11.0.12+7, mixed mode)

And 

> java version "1.8.0_371"
> Java(TM) SE Runtime Environment (build 1.8.0_371-b11)
> Java HotSpot(TM) 64-Bit Server VM (build 25.371-b11, mixed mode)

Both come up with the same error, nothing to find in existing issues

alorbach, 2023-06-14:
Opened jar file with Winrar, no error at all so the file is valid: 

[image]

SheharyarSMAdil, 2023-06-17:
@alorbach Facing same issue.

alorbach, 2023-06-21:
@me-no-dev I think this just needs a new rebuild with updated build tools, Would be great if you could update the release files thanks.