me-no-dev/EspExceptionDecoder #89: Esp8266 - ERROR: xtensa-lx106-elf-gdb.exe not found!

AndreiKorchunov, 2023-08-03:
Arduino IDE 1.8.19
Boards:
AVR 1.8.3
Esp32 2.0.11
Esp8266 3.1.2
Esp exception decoder 2.0.3

When I try to run the esp exception decoder, I get the following error:

```
gdbPath: C:\Users\kors\Documents\ArduinoData\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823/bin/xtensa-lx106-elf-gdb.exe

ERROR: xtensa-lx106-elf-gdb.exe not found!
```
It can be seen that the problem is that the search for xtensa-lx106-elf-gdb.exe for some reason is carried out along the path C:\Users\kors\Documents\ArduinoData\packages\esp32, although the file itself is located in c:\Users\kors \Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\bin\xtensa-lx106-elf-gdb.exe

Esp exception decoder is installed as recommended by the developer:
c:\Arduino\tools\EspExceptionDecoder\tool\EspExceptionDecoder.jar

The suggested solution https://github.com/me-no-dev/EspExceptionDecoder/issues/36#issuecomment-465846882 above didn't help.
Just in case, I reinstalled c:\Users\kors\Documents\ArduinoData\packages\esp32. Did not help
How to fix?

Peterkn2001, 2023-08-11:
It seems that the later versions of the ESP32 core use a different file naming convention, which is causing the error.
Annoying if you're trying to use the exception decoder for an ESP8266.

The workaround is to temporarily downgrade your ESP32 core to an earlier version (I randomly chose v2.0.5 and that works), but that obviously isn't much use if you're trying to decode an ESP32 exception report.

AndreiKorchunov, 2023-08-20:
It is not clear why the search is generally on the esp32 branch, although I have esp8266 selected

AndreiKorchunov, 2023-08-28:
Forgot to mention: used device Esp8266 - Node MCU 1.0 (Esp-12E module)

brendanmatkin, 2023-10-19:
A slightly easier workaround - I manually copied the file from the esp8266 core:
 For me: `...\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\bin\xtensa-lx106-elf-gdb.exe` 
 into the edbPath printed out in the error
 For me: `...\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823\bin`

AndreiKorchunov, 2023-10-19:
Thank you. As an option - it will do. But I would like it to work correctly :)

sgfslxf, 2023-11-14:
The problem was solved when I deleted the file C:\Users\**\Documents\ArduinoData\packages\esp32 and deleted the esp8266 file, then reinstalled the esp8266, version 2.7

rin1024, 2024-11-12:
I got the same problem(v2.0.3) and downgraded to 2.0.2, it works fine for me.

```
Arduino IDE: 1.18.19
Chip: Apple M1 Max
OS: 15.0.1 (24A348)
```