me-no-dev/EspExceptionDecoder #91: Arduino IDE 2.0.3

Abhay-2412, 2023-09-22:
"I downloaded the `'EspExceptionDecoder-2.0.3'` folder from your releases and followed the instructions by creating a 'tools' folder in my Arduino Sketch Directory. Specifically, I placed the folder at the following path: 

`'C:\Users\37175\Documents\Arduino\tools\EspExceptionDecoder-2.0.3\EspExceptionDecoder\tool\EspExceptionDecoder.jar'. `

However, after restarting the Arduino IDE, I couldn't find the exception decoder option. Could someone please kindly assist me with this issue?"

vicatcu, 2023-10-04:
Arduino 2.0.x changed from using Java to using the Electron framework, and the method of providing / launching external tools has changed / hasn't been figured out yet is my understanding.

3tcubed, 2023-11-14:
UP to IDE 2.2.1 and still no tools to Create FFAT partations so files can be uploaded, also preventing OTA updates from working when using IDE 2.2.1 --  Arrgh, I now waited nearly a year.  Time to learn PlatformIO and VSCode?  So much is still missing, I'm afraid you made a wrong turn in moving to Electron if these deficiencies can't be corrected/addressed an a year.

Oleg-Perevyshin, 2023-11-29:
**Maybe someone knows where to look at the development plan?**

I really need to work with the file system for ESP32 (Littlefs).
If there are no plans, then I will go out with Arduino.

cc46808, 2024-02-11:
> "I downloaded the `'EspExceptionDecoder-2.0.3'` folder from your releases and followed the instructions by creating a 'tools' folder in my Arduino Sketch Directory. Specifically, I placed the folder [...]

@Abhay-2412 I was able use this fork of EspExectionDecoder in ArduinoIDE 2.3.0

[https://github.com/dankeboy36/esp-exception-decoder?tab=readme-ov-file](https://github.com/dankeboy36/esp-exception-decoder?tab=readme-ov-file)

[image]

Doppenberg, 2024-02-15:
> > "I downloaded the `'EspExceptionDecoder-2.0.3'` folder from your releases and followed the instructions by creating a 'tools' folder in my Arduino Sketch Directory. Specifically, I placed the folder at the following path:
> > `'C:\Users\37175\Documents\Arduino\tools\EspExceptionDecoder-2.0.3\EspExceptionDecoder\tool\EspExceptionDecoder.jar'. `
> > However, after restarting the Arduino IDE, I couldn't find the exception decoder option. Could someone please kindly assist me with this issue?"
> 
> @Abhay-2412 I was able use this fork of EspExectionDecoder in ArduinoIDE 2.3.0
> 
> https://github.com/dankeboy36/esp-exception-decoder?tab=readme-ov-file
> 
> ![Screenshot 2024-02-11 143225](https://private-user-images.githubusercontent.com/16018709/303942896-cdb9c869-67cc-48f2-b47b-7193b49825d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5OTIyODEsIm5iZiI6MTcwNzk5MTk4MSwicGF0aCI6Ii8xNjAxODcwOS8zMDM5NDI4OTYtY2RiOWM4NjktNjdjYy00OGYyLWI0N2ItNzE5M2I0OTgyNWQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE1VDEwMTMwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjNGVhYjlkYTZjZWI4NzQ1Njc4ZmRkNjQzZmFlODAxNTBiMzA4MWY0OWNmYmY4MzVhN2RkYjVkODdkMzk4MDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CPRF8wOSlWxZcJwGCGYoj1RW4JFuplkdZyVDY9KIjh4)

Thank you! That fork you referenced worked well in Arduino IDE 2.3.0

IanBUK, 2024-12-31:
I've found I have to have Arduino IDE 1 and 2 installed, as this exception tool only works with 1.X.